analogdevicesinc/libiio #653: Find a way to handle the 30 seconds timeout when closing a serial port connection

dNechita, 2021-02-26:
Relevant information:
PR:  #652 where the problem is decribed

pcercuei, 2021-04-09:
@dNechita the corresponding PR was merged, should this issue be closed?

rgetz, 2021-04-24:
things get locked - just later...

right now, there is no async communication in libiio. when a scan or create_context happens, things return when things are done. by putting the uart open/close in a different thread - it will just wait for the thread to close by 30 seconds - it doesn't speed up the serial driver; and doesn't speed up when the thread is joined, so the application still sees a 30 second delay. The alternative is to return to the application, with the close thread still running, where someone can re-open the port with a different setting (baud rate), and experience the delay there. There is no good / robust answer (that I can think of).

see https://stackoverflow.com/questions/14255314/close-function-on-serial-port-descriptor-blocks-in-linux

rgetz, 2022-02-17:
Closing - since there is no way to do this in userspace.