analogdevicesinc/libiio #694: cmake: add uninstall target

willcode, 2021-06-02:
An "uninstall" target is useful both for development and for source-based package management (PyBOMBS in our case).

While cmake does not have a convenient uninstall, it is typically implemented using code like:
https://gist.github.com/royvandam/3033428

rgetz, 2022-02-22:
Thanks - we will have a look - and try to add

rgetz, 2023-02-08:
There should be file "install_manifest.txt" after executing `make install`. Either of the following commands should work:

`cat install_manifest.txt | sudo xargs rm`

or 

``xargs sudo rm < install_manifest.txt``

Does that solve your issue?

willcode, 2023-02-09:
Hi @rgetz, sure that's fine for a manual uninstall. I put this in originally because PyBOMBS runs `make uninstall` explicitly. This issue has been up for quite a while, and I don't hear any complaints. Not that surpising since the PyBOMBS recipe calls for an ancient tag. Feel free to close it if adding `make uninstall` is not a priority.

The recipe has specified v0.21 for a while now. I'd guess that people using PyBOMBS and PLUTO are installing via a package manager.

rgetz, 2023-05-06:
When #961 gets merged - this can be closed.

rgetz, 2023-10-16:
this can be closed - it was included in 0.25

closed 2023-10-16 by willcode