analogdevicesinc/libiio #698: Simultaneous access to gyroscope and accelerometer

buemi, 2021-06-11:
Hi!
Is there any simple way to simultaneously grab the gyroscope and accelerometer data using the readdev code?
Commands as the following:
    iio_readdev -T 0 -b 1 -s 1000 lsm6dsr_accel
seems to be the only possible use.
Regards.

mhennerich, 2021-06-11:
`root@analog:~# iio_readdev --help                                                                                                                                                                          
Usage:                                                                                                                                                                                                     
        iio_readdev [-n <hostname>] [-t <trigger>] [-T <timeout-ms>] [-b <buffer-size>] [-s <samples>] <iio_device> [<channel> ...]                                                                        
                                                                                                                                                                                                           
Options:                                                                                                                                                                                                   
        -h, --help                                                                                                                                                                                         
                        Show this help and quit.                                                                                                                                                           
        -n, --network                                                                                                                                                                                      
                        Use the network backend with the provided hostname.                                                                                                                                
        -u, --uri                                                                                                                                                                                          
                        Use the context with the provided URI.                                                                                                                                             
        -t, --trigger                                                                                                                                                                                      
                        Use the specified trigger.                                                                                                                                                         
        -b, --buffer-size                                                                                                                                                                                  
                        Size of the capture buffer. Default is 256.                                                                                                                                        
        -s, --samples                                                                                                                                                                                      
                        Number of samples to capture, 0 = infinite. Default is 0.                                                                                                                          
        -T, --timeout                                                                                                                                                                                      
                        Buffer timeout in milliseconds. 0 = no timeout                                                                                                                                     
        -a, --auto                                                                                                                                                                                         
                        Scan for available contexts and if only one is available use it.                                                                                                                   
root@analog:~#`

iio_readdev takes as last argument a list of channels you want to enable/capture

buemi, 2021-06-11:
> ``` [...]

Sorry, let me understand.
In my case I see the accelerometer and the gyroscope as different devices (I use "lsm6dsr_accel" and "lsm6dsr_gyro" as last argument to discern them). It seems that I can access to multiple channels of a single device, while I would like to access to both the devices.

mhennerich, 2021-06-11:
IIO knows only one buffer per device. This tool is not magically resolving this principle.
And in fact it can't be solved since the two devices may sample with different frequencies, etc.

buemi, 2021-06-11:
Ok,
thank you.
So in my case, is not possible to grab simultaneously the data from gyro and accel, even if I set them to the same frequency. Isn't it?

mhennerich, 2021-06-11:
Yes

closed 2021-08-11 by pcercuei