analogdevicesinc/libiio #705: Windows on ARM, Support?

sidd-kishan, 2021-06-20:
Please let us know when can we have an ARM64 version for Windows on ARM OS. We can help you test We have Windows on Rasberry Pi setup. Please pursue it we at Windows on Rasberry Pi community will be glad to extend support in testing your drivers and tools for ARM64.

Yamakaja, 2021-06-21:
Hey, just to clarify. Are you asking for:

* libiio support on Windows / ARM64, to access remote sensors using the ip / serial / usb backends

or

* Support for the "local" backend, i.e. enabling (lib)iio to access local sensors on your Raspberry PI from Windows?

sidd-kishan, 2021-06-21:
Dear @Yamakaja ,
                             Please provide us support for:-

`libiio support on Windows / ARM64, to access remote sensors using the ip / serial / usb backends`

so that we can interface analog devices with windows / ARM64 over ip/serial/usb backends and possibly develop instrumentation around libiio on Windows / ARM64

Yamakaja, 2021-06-21:
I can't really test this myself because i don't have an appropriate device ready, but have you tried building libiio from source on the device? Where does it fail?

sidd-kishan, 2021-06-21:
I have windows on ARM running on QEMU and have Windows 10 running on Rasberry pi the ARM64 version. I haven't tried building it but if you provide me a build I can test it out on multiple platforms.

pcercuei, 2021-06-23:
Libiio most likely already works on Windows ARM64. We just don't ship a binary (yet) for it, so you'd have to compile it yourself.

rgetz, 2021-11-08:
@tfcollins is this (Win/Arm64) something that Azure supports?

rgetz, 2022-02-17:
Yeah, this was asked for : https://developercommunity.visualstudio.com/t/please-add-a-windows-arm64-hosted-agent/1015752 but it seems to go unanswered...

rgetz, 2022-02-22:
I will close this, until upstream/prerequisites support this.

closed 2022-02-22 by rgetz