analogdevicesinc/libiio #708: Bug with iio_push_partial and Adam Pluto SDR

RGerzaguet, 2021-06-29:
Hello
I encounter a bug with libIIO and AD9361 board (with use of Adalm pluto device). To reproduce the bug, you can use the example C file `ad9361-iiostream.c`  (see [here](https://github.com/analogdevicesinc/libiio/blob/master/examples/ad9361-iiostream.c)) with a ADALM-Pluto device. The code run flawlessly with the Pluto SDR both in Tx and Rx modes. The code is run from a PC with Pluto address specification (i.e `./ad9361-iiostream usb:20.20.5`) 
            
- In the example provided, to transmit data it uses `nbytes_tx = iio_buffer_push(txbuf);` (see line 265). It transmits all the samples (1024x1024) (and it works)
- Now; if I want to transmit only a part of the Tx buffer, the previous line has to be modified with  for instance `nbytes_tx = iio_buffer_push_partial(txbuf,1024);`.
==> In this case I get an error 27 (`/* File too large */`) and then an error 9 (` Bad file descriptor`)

            ./ad9361-iiostream usb:20.20.5
                * Acquiring IIO context
                * Acquiring AD9361 streaming devices
                * Configuring AD9361 for streaming
                * Acquiring AD9361 phy channel 0
                * Acquiring AD9361 RX lo channel
                * Acquiring AD9361 phy channel 0
                * Acquiring AD9361 TX lo channel
                * Initializing AD9361 IIO streaming channels
                * Enabling IIO streaming channels
                * Creating non-cyclic IIO buffers with 1 MiS
                * Starting IO streaming (press CTRL+C to cancel)
                Error pushing buf -27
                        RX     1.05 MSmp, TX 18446744073709.55 MSmp
                Error pushing buf -9

How can I use `iio_buffer_push_partial` ?

pcercuei, 2021-06-29:
Did you change the buffer size? Because `iio_buffer_push_partial(txbuf, 1024)` works perfectly fine here (with a Pluto).

RGerzaguet, 2021-06-30:
Thanks for the quick reply, it gave me some hits and I think I have figured it out.
I have used the Pluto with Matlab hardware support package and it has updated the Pluto firmware. I think (?) that it indeed changes the internal buffer size.
So switching back to the *legacy* firmware ([see the Wiki](https://wiki.analog.com/university/tools/pluto/users/firmware) here) solves the issue

Many thanks again !
BR
@+R

closed 2021-06-30 by RGerzaguet

pcercuei, 2021-06-30:
@RGerzaguet Do you know which version of the firmware was failing?

RGerzaguet, 2021-06-30:
@pcercuei It seems to be version 0.31