analogdevicesinc/libiio #711: Add libffi-dev to Linux dependencies

calumkenny, 2021-07-09:
Hello,

I encountered the following error running the final install command `sudo make install` in [the build instructions](https://github.com/analogdevicesinc/libiio/blob/master/README_BUILD.md):

'Package libffi was not found in the pkg-config search path.
      Perhaps you should add the directory containing `libffi.pc'
      to the PKG_CONFIG_PATH environment variable
      No package 'libffi' found'

I solved this using:

`sudo apt-get install libffi-dev`

I thought this may apply to other users so I highlight it here. Feel free to delete issue as appropriate if this doesn't apply. 

Cheers
C

tfcollins, 2021-07-09:
Can you provide the distribution, version, and architecture you are using so we can reproduce it?

calumkenny, 2021-07-09:
See details below:

Operating System: Debian GNU/Linux 10 (buster)
Kernel: Linux 4.9.202-g346c64fa
Architecture: arm

`uname -m` gives armv7l

calumkenny, 2021-07-12:
Hi, apologies, I've realised that this issue should have been raised on the [pyadi-iio](https://github.com/analogdevicesinc/pyadi-iio) instead of libiio, as the issue occurred on installation of the pyadi-iio library ([step 4 of Linux installation](https://wiki.analog.com/resources/tools-software/linux-software/pyadi-iio)); the libiio library installation was handled successfully without the libffi-dev error. 

It may still be worth updating the build instructions for libiio as this is a prerequisite for pyadi-iio.

tfcollins, 2021-07-16:
Closing since unrelated.

closed 2021-07-16 by tfcollins