analogdevicesinc/libiio #712: Installation of libiio on windows10

sdrlab, 2021-07-11:
Hey,
We've been trying to install libiio on windows 10 (using the installation launcher), and tried running one of the example files (ad9361-iiostream.c) through visual studio ide. But, it returns errors at compilation:

[image]

When we add (char*) at both missing places, it still returns errors:
[image]

Do you have any clue what we are doing wrong? Have others encountered the same problem?

What does the following image mean, does the library not work on windows:
[image]

Our goal is to write a c++ which runs the FPGA and the ad9361 configurations and can accept data and send data with our own arm design.

We've also tried another approach after we encountered a tutorial to run libiio on visual studio here:
https://wiki.analog.com/resources/tools-software/linux-software/building_libiio_for_windows
But we were not able to complete it as the file xmllint.exe was missing in the zip folder.

[image]

Thanks a lot

tfcollins, 2021-07-12:
Are you linking gains libiio?

pcercuei, 2021-10-05:
Closing this issue now, since there hasn't been a response for almost 3 months.

closed 2021-10-05 by pcercuei