analogdevicesinc/libiio #730: You deleted the tag for 0.22

dvzrv, 2021-08-10:
Hi! I package libiio for Arch Linux.

It seems you have just deleted the tag for 0.22 from this repository.
Could you elaborate on why?

Yesterday I have already [initiated the update of the package](https://github.com/archlinux/svntogit-community/commit/9c99896458b9fd418ccd8f2a408eb01e62ddc967#diff-3e341d2d9c67be01819b25b25d5e53ea3cdf3a38d28846cda85a195eb9b7203a) and created [a TODO for its dependants to be rebuilt](https://archlinux.org/todo/libiio-022/) (as we had issues with minor version updates in the past).

Please do not delete tags, but ideally just create a new one.

pcercuei, 2021-08-11:
Hi David,

Sorry about that. I was about to create a release then some late changes had to be included, so I removed the tag. I had no idea it would create such a mess.

Is it OK if I push it again but on a different commit?

Thanks for your work packaging it for Arch, by the way.

dvzrv, 2021-08-11:
> Is it OK if I push it again but on a different commit?

The cleaner solution would be: Push the tag for the same commit again and tag 0.22.1 for the last minute changes added on top. :)

This makes the process transparent, as from the outside it is not possible to even tell whether someone has "moved" a tag ([github does not have a protected tags feature](https://github.community/t/feature-request-protected-tags/1742)).
Users building from a moved tag would not notice they are building something else entirely. Users building from the auto-generated tarball will have a broken build, because the generated tarball will have a different checksum (as it is created for a different commit).

pcercuei, 2021-08-11:
I don't remember exactly on what commit was the v0.22 tag though... I assume it was at e2ba5d48 but the .tar.gz for this hash downloaded from Github doesn't give me the same SHA512 as you.

mhennerich, 2021-08-11:
I think the v0.22 tag was created twice...

pcercuei, 2021-08-11:
We never force-push the master branch though, so it was pointing to one of the commits since e2ba5d4.

icchalmers, 2021-08-11:
> I don't remember exactly on what commit was the v0.22 tag though... I assume it was at [e2ba5d4](https://github.com/analogdevicesinc/libiio/commit/e2ba5d485bc7890c914c24a32e606a8d47b1bf4b) but the .tar.gz for this hash downloaded from Github doesn't give me the same SHA512 as you.

I can confirm that when I was updating my yocto recipe, e2ba5d4 was the commit associated with the v0.22 tag.

pcercuei, 2021-08-11:
I re-created the v0.22 tag from e2ba5d4, please tell me if it works for you.

dvzrv, 2021-08-11:
@pcercuei Yes, thank you! The tarball remains with the same checksum :)

pcercuei, 2021-08-11:
Alright. Then we will tag the real v0.22 version (the one we will ship as a release) as `v0.22-r1` in a few days.

mhennerich, 2021-08-11:
or just release v0.23 ?

pcercuei, 2021-08-11:
@mhennerich or that, yes. That would mean not creating a release for v0.22.

dvzrv, 2021-08-11:
> Alright. Then we will tag the real v0.22 version (the one we will ship as a release) as v0.22-r1 in a few days.

FWIW, `0.22-r1` < `0.22` according to [semver](https://semver.org).
`0.22.1` would be correct (or `0.23` at your avail).

pcercuei, 2021-08-11:
Alright. Thanks for raising the issue here.

closed 2021-08-11 by pcercuei