analogdevicesinc/libiio #745: Performance while switching between buffered read of two channels

magnuslundq, 2021-09-10:
Hello!

I am trying to read two channels using buffered read using the python bindings on an iMX6ULL CPU running a custom yocto built Linux 5.4.24.
While reading both channels interleaved with an interval of 0.05 seconds on each channel the CPU utilization rises to over 40%.
My understanding is that the buffer has to be deleted and re-created each time the channel mask is changed and that the channels cannot be changed on an existing buffer. 

This is some example code which reproduces the problem. I hope that maybe I am just missing something, but even if I remove the buffer.refill() and buffer.read() calls, only creating the buffer utilizes over 30% CPU.

```
import iio
import asyncio
import os

def write_file(path, filename, value):
    filepath = os.path.join(path, filename)
    file_handle = open(filepath, "w")
    file_handle.write(str(value))
    file_handle.close()

def read_file(path, filename):
    file_handle = open(os.path.join(path, filename), "r")
    value = file_handle.read()
    file_handle.close()
    return value

adc1_name = "2198000.adc"
adc1 = None
context_names = iio.scan_contexts()
for context_name in context_names:
    context = iio.Context(context_name)
    for dev in context.devices:
        if dev.name:
            if adc1_name in dev.name:
                adc1 = dev
try:
    iio_base = os.path.join(os.sep, "sys", "bus", "iio", "devices")
    write_file(os.path.join(iio_base, "iio_sysfs_trigger"), "remove_trigger", 1)
    write_file(os.path.join(iio_base, "iio_sysfs_trigger"), "add_trigger", 1)
    sysfs_trigger_path = os.path.join(iio_base, "trigger0")
    trigger_name = read_file(sysfs_trigger_path, "name")
    write_file(os.path.join(iio_base, adc1.id, "trigger"), "current_trigger", trigger_name)
except:
    pass

for channel in adc1.channels:
    channel.enabled = False

channel_1 = adc1.channels[6]
channel_2 = adc1.channels[9]

channel_1.attrs["sampling_frequency"]._write("29676")
channel_2.attrs["sampling_frequency"]._write("29676")

samples_count = 1

async def read_adc(channel):
    channel.enabled = True
    adc1_buffer = iio.Buffer(adc1, samples_count, cyclic=False)
    adc1_buffer.set_blocking_mode(True)
    adc1_buffer.refill()
    values = adc1_buffer.read()
    del adc1_buffer
    channel.enabled = False
    return values

async def sample_channel(channel, interval):
    while True:
        values = await read_adc(channel)
        await asyncio.sleep(interval)

async def run():
    await asyncio.gather(sample_channel(channel_1, 0.05),
                         sample_channel(channel_2, 0.05))

asyncio.run(sample_channels())
```

Any help and feedback is much appreciated.

pcercuei, 2021-09-16:
Hi @magnuslundq,

As you can see - creating and destroying buffer objects in a loop is a bad idea.

Enable both your channels 1 & 2, then create one iio.Buffer. When calling refill() you are then sampling both channels at the same time, and you can use iio.Channel.read() to extract each channel's data.

Cheers,
-Paul

rgetz, 2022-02-17:
Closing - since I think Paul's answer covers it.

closed 2022-02-17 by rgetz