analogdevicesinc/libiio #749: README_BUILD.md error, no Makefile.

jpnielsen, 2021-10-03:
"make" doens't work for a cmake project.

I had to do:

cmake ../ -DPYTHON_BINDINGS=OFF -DENABLE_IPV6=OFF
cmake --build .

tfcollins, 2021-10-03:
make is the primary tool we use and many projects use to build libraries once configured by cmake. Why do you say it doesn't work?

jpnielsen, 2021-10-03:
I'm trying to follow the build guide i found here: https://github.com/analogdevicesinc/libiio/blob/master/README_BUILD.md

this is what I get:

```
$ cd ~

jpn99@DESKTOP-KNC1KMC MINGW64 ~
$ git clone https://github.com/analogdevicesinc/libiio.git
Cloning into 'libiio'...
remote: Enumerating objects: 14517, done.
remote: Counting objects: 100% (2219/2219), done.
remote: Compressing objects: 100% (1020/1020), done.
remote: Total 14517 (delta 1317), reused 1604 (delta 1076), pack-reused 12298
Receiving objects: 100% (14517/14517), 14.64 MiB | 24.34 MiB/s, done.
Resolving deltas: 100% (9370/9370), done.

jpn99@DESKTOP-KNC1KMC MINGW64 ~
$ cd libiio

jpn99@DESKTOP-KNC1KMC MINGW64 ~/libiio
$ mkdir build

jpn99@DESKTOP-KNC1KMC MINGW64 ~/libiio
$ cd build

jpn99@DESKTOP-KNC1KMC MINGW64 ~/libiio/build
$ cmake ../ -DPYTHON_BINDINGS=ON
-- Building for: Ninja
-- cmake version: 3.21.3
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- The C compiler identification is GNU 10.3.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/msys64/mingw64/bin/cc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Performing Test HAS_WSHADOW
-- Performing Test HAS_WSHADOW - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strndup
-- Looking for strndup - not found
-- Looking for strerror_r
-- Looking for strerror_r - not found
-- Looking for newlocale
-- Looking for newlocale - not found
-- Looking for in6addr_any
-- Looking for in6addr_any - not found
CMake Error at CMakeLists.txt:139 (message):
  IPv6 is not available in your system.

-- Looking for libusb-1.0 : Found
-- Looking for libusb_get_version
-- Looking for libusb_get_version - found
-- Found Git: C:/msys64/usr/bin/git.exe (found version "2.33.0")
-- Building with Network back end support
-- Building with mdns, A Public domain mDNS/DNS-SD library in C
CMake Deprecation Warning at tests/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

CMake Deprecation Warning at bindings/python/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- bin= lib= inc=
-- Found Python: C:/msys64/mingw64/bin/python3.9.exe (found version "3.9.7") found components: Interpreter
-- new
-- Python_EXECUTABLE C:/msys64/mingw64/bin/python3.9.exe
-- Found Python: Building bindings
-- Configuring incomplete, errors occurred!
See also "C:/msys64/home/jpn99/libiio/build/CMakeFiles/CMakeOutput.log".
See also "C:/msys64/home/jpn99/libiio/build/CMakeFiles/CMakeError.log".

jpn99@DESKTOP-KNC1KMC MINGW64 ~/libiio/build
$ cmake ../ -DPYTHON_BINDINGS=OFF -DENABLE_IPV6=OFF
-- cmake version: 3.21.3
CMake Deprecation Warning at CMakeLists.txt:2 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Looking for libusb-1.0 : Found
-- Building with Network back end support
-- Building with mdns, A Public domain mDNS/DNS-SD library in C
CMake Deprecation Warning at tests/CMakeLists.txt:1 (cmake_minimum_required):
  Compatibility with CMake < 2.8.12 will be removed from a future version of
  CMake.

  Update the VERSION argument <min> value or use a ...<max> suffix to tell
  CMake that the project does not need compatibility with older versions.

-- Configuring done
-- Generating done
-- Build files have been written to: C:/msys64/home/jpn99/libiio/build

jpn99@DESKTOP-KNC1KMC MINGW64 ~/libiio/build
$ make -j$(nproc)
make: *** No targets specified and no makefile found.  Stop.
```

tfcollins, 2021-10-03:
> jpn99@DESKTOP-KNC1KMC MINGW64 ~/libiio/build
> $ cmake ../ -DPYTHON_BINDINGS=ON
> -- Building for: Ninja
> -- cmake version: 3.21.3

You've configured your build system to use Ninja and not makefiles. Hence why the make command tells you there are no make targets. I don't really use Ninja myself but I believe replacing **make** with **ninja** would effectively do the same thing in the build instructions.

jpnielsen, 2021-10-03:
yes, Thank you. I learned something-

closed 2021-10-03 by jpnielsen