analogdevicesinc/libiio #757: Build failure on v0.23

jo7ueb, 2021-11-02:
Hi,

I tried to build v0.23 but failed.
Something about libusb is not working correctly.
How can I fix it?

```
yasu@Dahdit-LinuxTest:~/dahdit/tombi/oss/libiio$ cmake .
-- cmake version: 3.16.3
-- Looking for libusb-1.0 : Found
-- Building with Network back end support
-- Building with Avahi, a DNS-SD (ZeroConf) implementation
-- Looking for libaio: Found
-- Configuring done
-- Generating done
-- Build files have been written to: /home/yasu/dahdit/tombi/oss/libiio
yasu@Dahdit-LinuxTest:~/dahdit/tombi/oss/libiio$ make
Scanning dependencies of target iio
[  2%] Building C object CMakeFiles/iio.dir/backend.c.o
[  4%] Building C object CMakeFiles/iio.dir/channel.c.o
[  6%] Building C object CMakeFiles/iio.dir/device.c.o
[  8%] Building C object CMakeFiles/iio.dir/context.c.o
[ 10%] Building C object CMakeFiles/iio.dir/buffer.c.o
[ 12%] Building C object CMakeFiles/iio.dir/utilities.c.o
[ 14%] Building C object CMakeFiles/iio.dir/scan.c.o
[ 16%] Building C object CMakeFiles/iio.dir/sort.c.o
[ 18%] Building C object CMakeFiles/iio.dir/usb.c.o
[ 20%] Building C object CMakeFiles/iio.dir/local.c.o
[ 22%] Building C object CMakeFiles/iio.dir/network.c.o
[ 25%] Building C object CMakeFiles/iio.dir/network-unix.c.o
[ 27%] Building C object CMakeFiles/iio.dir/dns_sd_avahi.c.o
[ 29%] Building C object CMakeFiles/iio.dir/dns_sd.c.o
[ 31%] Building C object CMakeFiles/iio.dir/xml.c.o
[ 33%] Building C object CMakeFiles/iio.dir/lock.c.o
[ 35%] Building C object CMakeFiles/iio.dir/iiod-client.c.o
[ 37%] Linking C shared library libiio.so
[ 39%] Built target iio
Scanning dependencies of target iio_tests_helper
[ 41%] Building C object tests/CMakeFiles/iio_tests_helper.dir/iio_common.c.o
[ 43%] Linking C static library libiio_tests_helper.a
[ 45%] Built target iio_tests_helper
[ 47%] Linking C executable iio_writedev
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_free_config_descriptor'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_open'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_get_device_list'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_control_transfer'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_exit'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_free_device_list'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_cancel_transfer'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_get_string_descriptor_ascii'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_free_transfer'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_submit_transfer'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_get_device_descriptor'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_claim_interface'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_handle_events_completed'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_close'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_set_auto_detach_kernel_driver'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_get_bus_number'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_init'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_get_device_address'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_alloc_transfer'
/usr/bin/ld: ../libiio.so.0.23: undefined reference to `libusb_get_active_config_descriptor'
collect2: error: ld returned 1 exit status
make[2]: *** [tests/CMakeFiles/iio_writedev.dir/build.make:87: tests/iio_writedev] Error 1
make[1]: *** [CMakeFiles/Makefile2:230: tests/CMakeFiles/iio_writedev.dir/all] Error 2
make: *** [Makefile:130: all] Error 2
```

FYI: libusb is installed.
```
$ dpkg -l | grep libusb
ii  libgusb2:amd64                             0.3.4-0.1                             amd64        GLib wrapper around libusb1
ii  libusb-0.1-4:amd64                         2:0.1.12-32                           amd64        userspace USB programming library
ii  libusb-1.0-0:amd64                         2:1.0.23-2build1                      amd64        userspace USB programming library
ii  libusb-1.0-0-dev:amd64                     2:1.0.23-2build1                      amd64        userspace USB programming library development files
ii  libusb-1.0-doc                             2:1.0.23-2build1                      all          documentation for userspace USB programming
ii  libusb-dev                                 2:0.1.12-32                           amd64        userspace USB programming library development files
ii  libusbmuxd6:amd64                          2.0.1-2                               amd64        USB multiplexor daemon for iPhone and iPod Touch devices - library
```

tfcollins, 2021-11-02:
What Linux distribution and version are you on?

tfcollins, 2021-11-04:
This still an issue?

pcercuei, 2022-01-07:
No feedback in two months - closing issue.

closed 2022-01-07 by pcercuei