analogdevicesinc/libiio #772: Build failures on MSVC

reid-p, 2021-12-09:
With the pattern change from 
#if 	 WITH_USB_BACKEND
..
#endif
to 
  if (WITH_USB_BACKEND) {
    ....
  }

I'm getting linker errors related to usb_context_scan() etc.

MSVC compiler is not smart enough to do dead code removal, or I haven't found the right switches.
Been some other reports in for different libraries of the same issue with MSVC.

Thoughts?

rgetz, 2021-12-10:
my thought is don’t modify the source so that it breaks. Sorry - we aren’t general purpose MSVC support - it has issues , but is popular enough we work around them.

The code as is - works. On purpose. We do it that way - not because it is pretty, but because it’s functional. 

- Robin

reid-p, 2021-12-12:
G'day rgetz,

Sorry let me be more clear. I haven't modified the source.
With commit c10d6e32fba6eacae178e7c3f37c44111a25b6dc 
treewide: Avoid conditional compilation 

msvc 2019 fails to compile libiio. The is due to the pattern change.
#if WITH_USB_BACKEND
..
#endif
to
if (WITH_USB_BACKEND) {
....
}

Prior to this commit it builds and links ok for me.

rgetz, 2021-12-13:
Recent builds on Azure don't have problems.

MSVC 2019 64-bit:
https://dev.azure.com/AnalogDevices/OpenSource/_build/results?buildId=8122&view=logs&jobId=3bf45259-f282-5ab9-6af8-d5171104ab20&j=3bf45259-f282-5ab9-6af8-d5171104ab20&t=8bcefb13-031a-5d6c-f5c0-32b1d654e425

and 32-bit:
https://dev.azure.com/AnalogDevices/OpenSource/_build/results?buildId=8122&view=logs&jobId=ad766681-fedf-5bb6-6d2e-1115be53eaad&j=ad766681-fedf-5bb6-6d2e-1115be53eaad&t=d62301a8-0134-58d5-d20d-2758cb22f19d

Are you compiling with any weird flags that turn off optimization?

pcercuei, 2021-12-13:
Hi @reid-p,
I can reproduce it, this happens when building the library when the cmake build variant is set to `Debug`. It will succeed to compile if you set it to `Release` mode, or `RelWithDebInfo` (if you need the debug symbols).

reid-p, 2021-12-14:
Thanks @pcercuei 

Yeap debug builds are the problem with msvc builds failing.

I was trying to create a vcpkg port for libiio for use with msvc. 
Which builds the debug libraries first, and that's what I also used for for independent testing.

Having debug builds has been nice occasionally.

pcercuei, 2021-12-14:
@reid-p The problem is that MSVC in debug mode is just very stupid. If it sees this code:
```
if (0)
   call_function();
```
It will try to link against the `call_function` symbol even though it's clearly dead code.

I believe for your vcpkg, you should build in `RelWithDebInfo` mode. The "solution" in our case would be to disable the `Debug` mode completely.

rgetz, 2022-02-17:
@pcercuei Can we disable ```Debug``` mode in MSVC?

pcercuei, 2022-02-17:
@rgetz I tried but failed, MSVC has a "Debug" mode even when CMake doesn't have one.

rgetz, 2022-02-17:
so, do we just document (in the ```README_BUILD.md```) and close?

rgetz, 2022-02-21:
Added #798

closed 2022-02-22 by rgetz