analogdevicesinc/libiio #776: C# Bindings ScanContext issue

Hagtronics, 2022-01-03:
Rev C Pluto, Latest Pluto Drivers (Dec 2021), Windows 10 Pro, Running Scan Context does not work for one instance as shown below,
         // Setup
         Dictionary<string, string> dic;
         iio.ScanContext whosThere = new iio.ScanContext();
         
         // Test code
         dic = whosThere.get_local_backend_contexts();  // Does not work - Shows zero devices

         dic = whosThere.get_usb_backend_contexts();    // works - shows connected device

         dic = whosThere.get_dns_sd_backend_contexts(); // works - shows connected device

tfcollins, 2022-01-03:
This is expected. Local contexts only apply to systems with locally available IIO drivers. This will never return any contexts on a Windows since the Windows kernel does not implement the IIO framework.

Hagtronics, 2022-01-03:
Thanks for the quick response Travis. Closed.

closed 2022-01-03 by Hagtronics