analogdevicesinc/libiio #794: iio_scan_context_get_info_list Generating errors when no devices available

tfcollins, 2022-02-18:
This appeared within that last few days.

Running iio_info and python equivalent returns errors on scanning. Example with fresh ubuntu 20.04 docker image:

iio_info
```
root@7ca8be9d8f5f:/work# iio_info -S
Library version: 0.23 (git tag: cad8314)
Compiled with backends: local xml ip usb
Scanning for IIO contexts failed: No such device (19)
```
Python
```
root@7ca8be9d8f5f:/work# python3 -c "import iio; ctxs = iio.scan_contexts()"
Traceback (most recent call last):
  File "<string>", line 1, in <module>
  File "/usr/local/lib/python3.8/dist-packages/iio.py", line 1511, in scan_contexts
    ctx_nb = _get_context_info_list(ctx, _byref(ptr))
  File "/usr/local/lib/python3.8/dist-packages/iio.py", line 62, in _check_negative
    raise OSError(-result, _strerror(-result))
```

pcercuei, 2022-02-18:
@tfcollins I can't reproduce. I built it from the exact same git commit, and it works fine on my ZedBoard (detects the local backend) and on my PC (detects my Pluto).

tfcollins, 2022-02-18:
Try it without a device or context available

pcercuei, 2022-02-18:
@tfcollins could you share the generated `iio-config.h` file?

tfcollins, 2022-02-18:
```
#ifndef IIO_CONFIG_H
#define IIO_CONFIG_H

#define LIBIIO_VERSION_MAJOR	0
#define LIBIIO_VERSION_MINOR	23
#define LIBIIO_VERSION_GIT	"cad8314"

#define LOG_LEVEL Info_L

#define WITH_LOCAL_BACKEND 1
#define WITH_XML_BACKEND 1
#define WITH_NETWORK_BACKEND 1
#define WITH_USB_BACKEND 1
#define WITH_SERIAL_BACKEND 0

/* #undef WITH_NETWORK_GET_BUFFER */
#define WITH_NETWORK_EVENTFD 1
#define WITH_IIOD_USBD 1
#define WITH_IIOD_SERIAL 1
#define WITH_LOCAL_CONFIG 1
#define WITH_LOCAL_MMAP_API 1
#define WITH_HWMON 0
#define WITH_AIO 1
#define HAVE_DNS_SD 0
#define HAVE_AVAHI 0
#define WITH_ZSTD 0

/* #undef HAS_PIPE2 */
#define HAS_STRDUP
#define HAS_STRNDUP
#define HAS_STRTOK_R
#define HAS_STRERROR_R
#define HAS_NEWLOCALE
#define HAS_PTHREAD_SETNAME_NP
#define HAVE_IPV6
/* #undef NO_THREADS */
#define HAS_LIBUSB_GETVERSION

#endif /* IIO_CONFIG_H */
```

rgetz, 2022-02-20:
yeah, this happens because we tell you to find something on ip, and ip is built without DNS_SD support...

So, this fixes it - I can make a PR if @pcercuei agrees with the fix.

```
diff --git a/scan.c b/scan.c
index 503864e..8030fba 100644
--- a/scan.c
+++ b/scan.c
@@ -155,7 +156,7 @@ struct iio_scan_context * iio_create_scan_context(
                return NULL;
        }
 
-       ctx->backendopts = iio_strndup(backend ? backend : "local,usb,ip", PATH_MAX);
+       ctx->backendopts = iio_strndup(backend ? backend : (HAVE_DNS_SD ? "local,usb,ip" : "local,usb"), PATH_MAX);
        if (!ctx->backendopts) {
                free(ctx);
                errno = ENOMEM;
```

pcercuei, 2022-02-20:
@rgetz that's not the right fix. If you disable the local or USB backends you will get the same issue.

rgetz, 2022-02-21:
yeah, this fixes things in a more general purpose way. 

Thanks

PR #795 (2022-02-21): scan: Fix default backends list containing disabled backends
Use CMake to give us a list of the backends available for scanning. This
makes sure that the context scan won't fail by trying to scan from a
backend that wasn't enabled.

Fixes #794.
commit 163f9a5f (2022-02-21): scan: Fix default backends list containing disabled backends

Use CMake to give us a list of the backends available for scanning. This
makes sure that the context scan won't fail by trying to scan from a
backend that wasn't enabled.

Fixes #794.
closed 2022-02-21 by rgetz