analogdevicesinc/libiio #797: Question: is libiio compatible with the latest kernel's support for output buffers?

allsey87, 2022-02-21:
It seems that support for output buffers has finally been merged into the mainline Linux kernel.[^1] However, it is unclear to me whether or not libiio is capable of using that support. 

I managed to get previous versions of libiio to work with older kernels by manually patching in support for output buffers based on work from Analog Device's fork of the kernel, however, the names and the signatures of the functions that landed in the mainline kernel differ a bit from this version, e.g., `iio_buffer_write_outer_addr` instead of `iio_buffer_chrdev_write`.

I have been trying to get to the bottom of whether the latest libiio supports the original API, the API that recently landed in the mainline, or perhaps whether libiio is agnostic to these changes? I have not been able to find an answer to this question.

[^1]: https://github.com/torvalds/linux/commit/9eeee3b0bf190b4f677af27e24ba0cd1c030e49b

pcercuei, 2022-02-22:
Hello Michael,

Yes, absolutely. Libiio will happily work with the regular read/write interface. It was supported since the inception of the library.

You are asking about the "original API" and the "API that recently landed", but those two are the same one. It's just that the downstream ADI kernel had write support way before it's been upstreamed.

ADI's kernel also has a "high-speed" interface where buffers of samples are passed back and forth between the kernel and libiio. But it is completely optional though and libiio will work happily without it.

One note on buffer write support - even if it's been recently wired in the IIO core, it is not yet supported upstream in the dmaengine backend (`drivers/iio/buffer/industrialio-buffer-{dma,dmaengine}.c`). I am working on that, so it should be upstream hopefully soon: https://lore.kernel.org/linux-doc/20220207125933.81634-7-paul@crapouillou.net/T/

allsey87, 2022-02-22:
Thanks for the reply and the information, @pcercuei! It's excellent to see this stuff finally making its way into the upstreamed kernel. I have been patching this stuff into kernels for the mobile robots that I work with for years!

pcercuei, 2022-02-22:
Sure, no problem. Don't hesitate to test and give feedback on the patchset I'm trying to upstream, it would be very helpful!

closed 2022-02-22 by pcercuei