analogdevicesinc/libiio #827: 0.23: despite build with `WITH_TESTS=ON` ctest is not able to find test units

kloczek, 2022-04-02:
Looks like something is wrong with test suite because despite run cmake with `WITH_TESTS=ON` ctest is not able to find test suite units
```console
[tkloczko@devel-g2v x86_64-redhat-linux-gnu]$ cmake -L| grep TEST
CMake Warning:
  No source or binary directory provided.  Both will be assumed to be the
  same as the current working directory, but note that this warning will
  become a fatal error in future CMake releases.

CMake Error: The source directory "/home/tkloczko/rpmbuild/BUILD/libiio-0.23/x86_64-redhat-linux-gnu" does not appear to contain CMakeLists.txt.
Specify --help for usage, or press the help button on the CMake GUI.
WITH_TESTS:BOOL=ON
[tkloczko@devel-g2v x86_64-redhat-linux-gnu]$ ctest -vv
Test project /home/tkloczko/rpmbuild/BUILD/libiio-0.23/x86_64-redhat-linux-gnu
No tests were found!!!
```

kloczek, 2022-04-04:
Why not jus hook usimg those tools into regulat ctest units? 🤔

pcercuei, 2022-04-04:
Well, these test programs are meant to be used with hardware, so it is not really easy to use them in unit tests...