analogdevicesinc/libiio #829: Using libiio.pc for MSVC

gvanem, 2022-04-06:
The name of the library becomes `libiio.lib` after Cmake + Nmake finishes.
But using the generated `libiio.pc` with this command `pkg-config libiio --libs --msvc-syntax`
gives:
  `/libpath:f:/ProgramFiler-x86/libiio/lib iio.lib`

There's no `iio.lib`. Hence pkg-config is unusable for MSVC.
Only works for MinGW: `pkg-config libiio --libs` correctly gives `-Lf:/ProgramFiler-x86/libiio/lib -liio`.

pcercuei, 2022-04-07:
Honestly, we don't use pkg-config on Windows because it's a pain.
However if you think this should be fixed, it's just a matter of setting the correct library name in `libiio.pc.cmakein` from one of the CMake variables.

rgetz, 2022-10-21:
@pcercuei , I don't know if that works...

man page:
```
--msvc-syntax
This option is available only on Windows. It causes pkg-config to output -l and -L flags in the form recognized
by the Microsoft Visual C++ command-line compiler, cl. Specifically, instead of 
-Lx:/some/path it prints /libpath:x/some/path, and instead of -lfoo it prints foo.lib. Note that the --libs output
consists of flags for the linker, and should be placed on the cl command line after a /link switch.
```

so I think it's doing exactly what it is supposed to - if you change the ```libiio.pc.cmakein``` - for MSVC - it will break for mingw.

Isn't the fix to take:
https://github.com/analogdevicesinc/libiio/blob/master/CMakeLists.txt#L469 and make it "iio.dll" only?

Thanks
-Robin