analogdevicesinc/libiio #830: [MinGW32] missing timespec_get()

gvanem, 2022-04-06:
Trying to build for MinGW with the minimal features possible:
```
cmake -G "MinGW Makefiles" -DWITH_XML_BACKEND=0 -DWITH_SERIAL_BACKEND=0 -DWITH_USB_BACKEND=0 -DENABLE_IPV6=0 -DPYTHON_BINDINGS=0 -DWITH_NETWORK_BACKEND=0 ..
```
Yields:
```
F:\gv\dx-radio\libiio\tests\iio_common.c: In function 'get_time_us':
F:\gv\dx-radio\libiio\tests\iio_common.c:455:2: warning: implicit declaration of function 'timespec_get'
[-Wimplicit-function-declaration]
  455 |  timespec_get(&tp, TIME_UTC);
      |  ^~~~~~~~~~~~
F:\gv\dx-radio\libiio\tests\iio_common.c:455:20: error: 'TIME_UTC' undeclared (first use in this function)
  455 |  timespec_get(&tp, TIME_UTC);
      |                    ^~~~~~~~
```

My MinGW (TDM / gcc version 10.3.0) do have an inlined `clock_gettime()` in it's `<pthread_time.h>` 
(which gets included in `<time.h>` for `_POSIX_C_SOURCE > 2`). 
But inlined only for `__MINGW64__`! So I tried this patch:
```diff
--- a/tests/iio_common.c 2022-04-04 10:17:44
+++ b/tests/iio_common.c 2022-04-06 10:59:46
@@ -451,7 +451,9 @@
 {
        struct timespec tp;

-#ifdef _WIN32
+#if defined(__MINGW32__)
+       __pthread_clock_gettime(CLOCK_REALTIME, &tp);
+#elif defined(_WIN32)
        timespec_get(&tp, TIME_UTC);
 #else
        clock_gettime(CLOCK_REALTIME, &tp);
```

The above compiles and links for both `-DBUILD_SHARED_LIBS=0` and `-DBUILD_SHARED_LIBS=1`.

pcercuei, 2022-04-06:
Since you say that your MinGW does have clock_gettime(), why not:
```
#if defined(_WIN32) && !defined(__MINGW32__)
    timespec_get(&tp, TIME_UTC);
#else
    clock_gettime(CLOCK_REALTIME, &tp);
#endif
```

gvanem, 2022-04-06:
But it is inlined only for `__MINGW64__` / `x64`; excluding 32-bit targets. 
But this issue is really not important for me. I've abandoned MinGW some years ago due to such messy issues.
So feel free to close or fix at own will.

gvanem, 2022-04-06:
Besides your `CMakeLists.txt` is really not for `x86` on `MINGW` (`-m32`) at all. 
How does this hellish Cmake system choose to build 32-bit targets?

pcercuei, 2022-04-06:
> Besides your `CMakeLists.txt` is really not for `x86` on `MINGW` (`-m32`) at all. How does this hellish Cmake system choose to build 32-bit targets?

Easy - just use a 32-bit MinGW toolchain :)

PR #835 (2022-04-06): tests: Fix build under MinGW
MinGW 32-bit does not have timespec_get(). Strangely, the 64-bit version
does have it. But both have clock_gettime(), so only use timespec_get()
when building under Visual Studio.

Fixes #830.
commit 47381e2d (2022-04-06): tests: Fix build under MinGW

MinGW 32-bit does not have timespec_get(). Strangely, the 64-bit version
does have it. But both have clock_gettime(), so only use timespec_get()
when building under Visual Studio.

Fixes #830.
closed 2022-04-06 by rgetz