analogdevicesinc/libiio #832: [MSVC] compiling examples/adrv9002-iiostream.c

gvanem, 2022-04-06:
The pre-processor in MSVC 2019 (ver. 19.29.30139) seems brain-dead. 
When compiling `examples/adrv9002-iiostream.c`, I get lots these errors:
```
examples/adrv9002-iiostream.c(130): error C2143: syntax error: missing ')' before 'string'
examples/adrv9002-iiostream.c(130): warning C4477: 'printf' : format string '%s' requires an argument 
of type 'char *', but variadic argument 1 has type '__int64'
...
```

In expansion of the macro `info("adrv9002 bandwidth: %lld\n", val);`. The pre-processed output really is:
```c
printf ("%s, %d: INFO: " "adrv9002 bandwidth: %lld\n", val"%s", __func__, 135, "");
```

But patching for `_MSC_VER` (but not clang-cl):
```diff
--- a/examples/adrv9002-iiostream.c 2022-04-04 10:17:44
+++ b/examples/adrv9002-iiostream.c 2022-04-06 11:14:23
@@ -14,6 +14,11 @@
 #include <errno.h>
 #include <stdbool.h>

+#if defined(_MSC_VER) && !defined(__clang__)
+  #define error(fmt, ...) printf("%s, %u: ERROR: " fmt, __func__, __LINE__, __VA_ARGS__)
+  #define info(fmt, ...)  printf("%s, %u: INFO: " fmt, __func__, __LINE__, __VA_ARGS__)
+
+#else
 #define ARGS(fmt, ...) __VA_ARGS__
 #define FMT(fmt, ...)  fmt
 #define error(...) \
@@ -21,6 +26,7 @@

 #define info(...) \
        printf("%s, %d: INFO: " FMT(__VA_ARGS__, 0)"%s", __func__, __LINE__, ARGS(__VA_ARGS__, ""))
+#endif
```

works for me: `main, 263: ERROR: Could not create IIO context`.

rgetz, 2022-04-06:
yeah, we really only compile / run the examples on Linux.

Do you need them on Windows?

gvanem, 2022-04-07:
@rgetz No, I do not need anything from IIO besides the `.lib/.dll` files really.
But a *"A cross platform library for interfacing with ..."* should be just that.

pcercuei, 2022-04-07:
The library itself is cross-platform, the examples are just that, examples. They aren't even built by default.

gvanem, 2022-04-07:
But why cannot gcc on Linux use this form too:
```c
 #define error(fmt, ...) printf("%s, %u: ERROR: " fmt, __func__, __LINE__, __VA_ARGS__)
```

I mean, what's the advantage of the current overly complex definitions?

pcercuei, 2022-04-07:
@gvanem with this macro definition, if you call just `error("foo\n");`, the compilation may fail, because the C standard dictates that there must be at least one argument in the `__VA_ARGS__` part. Hence the tricks used.

We are doing basically the same thing in the dev branch, and this builds fine under Visual Studio:
https://github.com/analogdevicesinc/libiio/blob/dev/include/iio/iio-debug.h#L63-L70

closed 2025-06-27 by dNechita