analogdevicesinc/libiio #845: iio_reg read bug using local backed

mhennerich, 2022-05-09:
Reading registers using iio_reg clobbers content when using the local backed:

```
root@analog:~# iio_reg axi-ad9172-hpc 0x8 0xc0
root@analog:~# iio_reg axi-ad9172-hpc 0x8
0x1
```

Doing the same using remote doesn't show this issue.

```
$ iio_reg -u ip:10.44.3.51 axi-ad9172-hpc 0x8 0xc0
$ iio_reg -u ip:10.44.3.51 axi-ad9172-hpc 0x8
0xc0
```
This is seen with the pre-release latest master:

```
root@analog:~# iio_info | head
Library version: 0.23 (git tag: 81b7802)
Compiled with backends: local xml ip usb
IIO context created with local backend.
Backend version: 0.23 (git tag: 81b7802)
Backend description string: Linux analog 5.10.0-98183-g2357ab1d0211-dirty #3715 SMP Thu May 5 17:33:07 CEST 2022 aarch64
```

mhennerich, 2022-05-09:
ok - the register I'm reading here is a page mask.
It needs to be set correctly for the following register read to return the paged value.

For some reason, using any of the tools, iio_attr, iio_reg, etc.
Will scan the context read some channel attributes...

Why is that?

In this case it'll read IIO_CHAN_INFO_SCALE which will clobber by page mask.
Doing this remote doesn't do this sort of scan context which appears to be destructive ....

root@analog:~# iio_reg -s
[  119.340929] ad9172_spi_xfer:552: 0x47, 0x0
[  131.734702] ad9172_read_raw:469: 0x2
[  131.739030] ad9172_spi_xfer:552: 0x8, 0x0
[  131.743179] ad9172_spi_xfer:552: 0x46, 0x0
[  131.747619] ad9172_spi_xfer:552: 0x47, 0x0
[  131.752027] ad9172_read_raw:469: 0x2
[  131.756828] ad9172_spi_xfer:552: 0x8, 0x0
[  131.760635] ad9172_spi_xfer:552: 0x46, 0x0

pcercuei, 2022-05-09:
Attributes like "scale" are read when constructing the local backend so that they appear in the XML string.

mhennerich, 2022-05-09:
Why? Scale may not be a constant...

pcercuei, 2022-05-09:
Well, I thought it was a constant back when the code was written back in 2014 :)

Looking at the ad9172 driver, you could probably work around this issue by using a cache for your regmap.

mhennerich, 2022-05-09:
Scale was always dynamic - the only constant should be name ...

What:		/sys/bus/iio/devices/iio:deviceX/in_accel_scale_available
What:		/sys/.../iio:deviceX/in_anglvel_scale_available
What:		/sys/.../iio:deviceX/in_magn_scale_available
What:		/sys/.../iio:deviceX/in_illuminance_scale_available
What:		/sys/.../iio:deviceX/in_intensity_scale_available
What:		/sys/.../iio:deviceX/in_proximity_scale_available
What:		/sys/.../iio:deviceX/in_voltageX_scale_available
What:		/sys/.../iio:deviceX/in_voltage-voltage_scale_available
What:		/sys/.../iio:deviceX/out_voltageX_scale_available
What:		/sys/.../iio:deviceX/out_altvoltageX_scale_available
What:		/sys/.../iio:deviceX/in_capacitance_scale_available
What:		/sys/.../iio:deviceX/in_pressure_scale_available
What:		/sys/.../iio:deviceX/in_pressureY_scale_available
KernelVersion:	2.6.35
Contact:	linux-iio@vger.kernel.org
Description:
		If a discrete set of scale values is available, they
		are listed in this attribute.

pcercuei, 2022-05-09:
Yes, I know, that's a mistake I made back then and we couldn't change it without breaking ABI.

pcercuei, 2022-05-09:
Btw - the scale encoded into the XML is only for scan-element channels, and it's encoded along with the other details on the data:
https://github.com/analogdevicesinc/libiio/blob/master/iio.h#L1668-L1695

You can still read the "scale" attribute as a regular channel attribute and it will be dynamic.

pcercuei, 2022-05-09:
Closing as this issue isn't really a bug in libiio, and will be resolved in the kernel.