analogdevicesinc/libiio #851: iio_attr negative value error for hardwaregain

ademeren, 2022-05-30:
If you try to give negative number for attenuation, software make a mistake and take the "-" symbol as an option. When I was searching, I saw the bug fixed. However, windows version of software still has the same error.

rgetz, 2022-09-13:
What version of libiio are you using on Windows?

rgetz, 2023-03-12:
I just tried this on windows 10:

```
C:\Users\rgetz>iio_info -S
Library version: 0.24 (git tag: c4498c2)
...

C:\Users\rgetz>iio_attr -u usb:3.1.5 -c -o ad9361-phy voltage0 hardwaregain
-10.000000 dB

C:\Users\rgetz>iio_attr -u usb:3.1.5 -c -o ad9361-phy voltage0 hardwaregain -6
-6.000000 dB
```

so I think this works, and this issue can be closed.

closed 2023-03-12 by tfcollins