analogdevicesinc/libiio #854: How to compile libiio on macOS M1 pure arm64, no Rosetta

altitudedashboard, 2022-06-17:
I could not find libiio on brew, it seems to be available on macports, but I'm not using macports and I'm unsure it's arm64

tfcollins, 2022-06-18:
There is a brew formula here: https://github.com/tfcollins/homebrew-formulae/blob/master/libiio.rb but it hasn't been tested on Apple Silicon

altitudedashboard, 2022-06-18:
I tried the following:

```
brew install cmake
brew install cmake-docs
brew install libserialport
brew install libusb

curl -LJO https://github.com/analogdevicesinc/libiio/archive/refs/tags/v0.23.zip
unzip libiio-0.23.zip
cd libiio-0.23
cmake .
make
sudo make install
```

but I get an error during make install:

```
sudo make install
Consolidate compiler generated dependencies of target iio
[ 47%] Built target iio
Consolidate compiler generated dependencies of target iio_tests_helper
[ 55%] Built target iio_tests_helper
Consolidate compiler generated dependencies of target iio_genxml
[ 61%] Built target iio_genxml
Consolidate compiler generated dependencies of target iio_info
[ 66%] Built target iio_info
Consolidate compiler generated dependencies of target iio_attr
[ 72%] Built target iio_attr
Consolidate compiler generated dependencies of target iio_readdev
[ 77%] Built target iio_readdev
Consolidate compiler generated dependencies of target iio_reg
[ 83%] Built target iio_reg
Consolidate compiler generated dependencies of target iio_writedev
[ 88%] Built target iio_writedev
Consolidate compiler generated dependencies of target iio_adi_xflow_check
[ 94%] Built target iio_adi_xflow_check
Consolidate compiler generated dependencies of target iio_stresstest
[100%] Built target iio_stresstest
Install the project...
-- Install configuration: "RelWithDebInfo"
-- Up-to-date: /usr/local/lib/pkgconfig/libiio.pc
-- Installing: /Library/Frameworks/iio.framework
-- Installing: /Library/Frameworks/iio.framework/Versions
-- Installing: /Library/Frameworks/iio.framework/Versions/0.23
-- Installing: /Library/Frameworks/iio.framework/Versions/0.23/Headers
-- Installing: /Library/Frameworks/iio.framework/Versions/0.23/Headers/iio.h
-- Installing: /Library/Frameworks/iio.framework/Versions/0.23/Resources
-- Installing: /Library/Frameworks/iio.framework/Versions/0.23/Resources/Info.plist
-- Installing: /Library/Frameworks/iio.framework/Versions/0.23/iio
-- Installing: /Library/Frameworks/iio.framework/Versions/Current
-- Installing: /Library/Frameworks/iio.framework/Headers
-- Installing: /Library/Frameworks/iio.framework/Resources
-- Installing: /Library/Frameworks/iio.framework/iio
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_genxml
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_info
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_attr
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_readdev
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_reg
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_writedev
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_adi_xflow_check
-- Installing: /Library/Frameworks/iio.framework/Tools/iio_stresstest
```

I'm sorry, I'm not really a developer. Just trying to make an ADALM-Pluto device work on macOS arm64 with gnuradio

altitudedashboard, 2022-06-18:
These commands failed:

```
curl -LJO https://github.com/tfcollins/homebrew-formulae/blob/master/libiio.rb
brew install --build-from-source ./libiio.rb
==> Tapping homebrew/cask
Cloning into '/opt/homebrew/Library/Taps/homebrew/homebrew-cask'...
remote: Enumerating objects: 647974, done.
remote: Counting objects: 100% (132/132), done.
remote: Compressing objects: 100% (53/53), done.
remote: Total 647974 (delta 84), reused 126 (delta 79), pack-reused 647842
Receiving objects: 100% (647974/647974), 306.62 MiB | 1.22 MiB/s, done.
Resolving deltas: 100% (458954/458954), done.
Tapped 4014 casks (4,086 files, 327.3MB).
Error: libiio: /Users/codrut/libiio.rb:8: syntax error, unexpected '<'
<!DOCTYPE html>
^
/Users/codrut/libiio.rb:9: syntax error, unexpected '<'
<html lang="en" data-color-mod...
^
/Users/codrut/libiio.rb:9: syntax error, unexpected tIDENTIFIER, expecting end-of-input
<html lang="en" data-color-mode="auto" data-light...
                ^~~~

```

altitudedashboard, 2022-06-18:
I'm sorry, I made some mistakes above. It seems this work:

```
curl -LJO https://raw.githubusercontent.com/tfcollins/homebrew-formulae/master/libiio.rb
brew install --build-from-source ./libiio.rb
```

closed 2022-06-18 by tfcollins