analogdevicesinc/libiio #859: How to use C# bindings

JoepBakker, 2022-07-03:
I am new to using libiio and I would like to use it in a csharp application. However, I have not been able to find any information about how to actually use/add the csharp bindings to my program.

Could someone maybe explain or link to a resource that explains how to set up a c# project that is able to use libiio functionality?

Thanks a lot in advance!

pcercuei, 2022-07-05:
Hi,

If you install libiio through the libiio-setup.exe then the C# bindings should already be installed. You can add it in Visual Studio (or your IDE of choice) like any other dependency.

If you wonder how to use the bindings, there is an (old) example here: https://github.com/analogdevicesinc/libiio/blob/master/bindings/csharp/examples/ExampleProgram.cs

Blackclaws, 2024-01-18:
It would be great if you could release a nuget package with the bindings and also the runtime as that is much easier to consume than installing libiio manually and also permits easy restore across applications. If you'd like someone to help with setting that up let me know.

dNechita, 2026-11-04:
The ability to create nuget packages was added in #1595.

closed 2026-11-04 by dNechita