analogdevicesinc/libiio #864: Automatically enumerate network devices

catkira, 2022-07-10:
Apps that use libiio do not automatically find network devices.
For example with iio_info it is necessary to specify for example
`iio_info -n 192.168.1.1` 

It would be nice to have a config file for libiio that contains known network devices so that apps automatically find them similar to usb connected devices.

tfcollins, 2022-07-10:
This already exists if you build with avahi. Just use iio_info -S

catkira, 2022-07-10:
I built libiio with avahi. However when I do iio_info -S i get
```
benja@DESKTOP:~$ iio_info -S
Library version: 0.23 (git tag: 3db19bc)
Compiled with backends: local xml ip usb
ERROR: Unable to create Avahi DNS-SD client :Daemon not running
Scanning for IIO contexts failed: Text file busy (26)
```
Does the avahi daemon also have to run on the PC? 
I can confirm that its running on Pluto. When I do iio_info -S on Pluto I get
```
# iio_info -S
Library version: 0.21 (git tag: v0.21)
Compiled with backends: local xml ip usb serial
Available contexts:
        0: (cf-ad9361-lpc, xadc, cf-ad9361-dds-core-lpc, ad9361-phy on Analog Devices PlutoSDR Rev.C (Z7010/AD9363)) [local:]
        1: 192.168.2.1 (Analog Devices PlutoSDR Rev.C (Z7010-AD9364)), serial= [ip:pluto.local]
        2: 192.168.137.2 (Analog Devices PlutoSDR Rev.C (Z7010-AD9364)), serial= [ip:pluto.local]
```

I think a problem could be, that my Pluto (Antsdr) has an ip (192.168.137.2) that is not part of the subnet that is used in my WSL linux
```
benja@DESKTOP:~$ ifconfig
eth0: flags=4163<UP,BROADCAST,RUNNING,MULTICAST>  mtu 1500
        inet 192.168.30.41  netmask 255.255.240.0  broadcast 192.168.31.255
        inet6 fe80::215:5dff:fe43:3edd  prefixlen 64  scopeid 0x20<link>
        ether 00:15:5d:43:3e:dd  txqueuelen 1000  (Ethernet)
        RX packets 49  bytes 9051 (9.0 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 14  bytes 1076 (1.0 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0

lo: flags=73<UP,LOOPBACK,RUNNING>  mtu 65536
        inet 127.0.0.1  netmask 255.0.0.0
        inet6 ::1  prefixlen 128  scopeid 0x10<host>
        loop  txqueuelen 1000  (Local Loopback)
        RX packets 0  bytes 0 (0.0 B)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 0  bytes 0 (0.0 B)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```

However iio_info -n 192.168.137.2 finds the device. Do You have any idea how I can make that work?
Starting avahi daemon on my PC does not work, but I suppose thats not needed right?
```
benja@DESKTOP:~$ sudo service avahi-daemon start
 * Starting Avahi mDNS/DNS-SD Daemon avahi-daemon                                                                [fail]
```

catkira, 2022-07-10:
I made it a bit further, my dbus service was not running. After I started dbus, I can also start avahi-daemon. When I do iio_info -S now I dont get an error, but it also does not find the device:
```
benja@DESKTOP:~$ iio_info -S
Library version: 0.23 (git tag: 3db19bc)
Compiled with backends: local xml ip usb
No IIO context found.
```

tfcollins, 2022-07-11:
Do you have NICs assigned in the subnets the target devices are within?

catkira, 2022-08-29:
The problem was that I am not using the RNDIS device directly in Linux. I run the RNDIS device in Windows 11 and run a Ubuntu 20 in WSL. The avahi data does not get transferred to the WSL, but using the ip directly in WSL works.

The problem does not seem to be inside libiio, so I clode this issue. Thank You!

closed 2022-08-29 by catkira