analogdevicesinc/libiio #867: Add cross compile feature to build script

catkira, 2022-07-14:
It would be nice if libiio could cross compile without buildroot. That would be very helpful when trying out new libiio versions. Some tiny systems like PlutoSDR dont have a GCC onboard, so libiio cant be compiled there.

pcercuei, 2022-07-18:
It can cross-compile with Buildroot. Just pass the `-DCMAKE_TOOLCHAIN_FILE=` option with the path to the Buildroot toolchain's share/buildroot/toolchainfile.cmake.

catkira, 2022-07-18:
ah ok thats nice. Thanks!
 I still dont have the toolchainfile.cmake because my buildroot build with vitis2022.1 is not working yet, but thats another issue.

catkira, 2022-08-03:
When I do this command
```
benja@DESKTOP:/mnt/d/git/libiio/build$ cmake .. -DCMAKE_TOOLCHAIN_FILE=/mnt/d/git/plutosdr-fw/buildroot/output/host/share/buildroot/toolchainfile.cmake
```
I get the following error
```
-- cmake version: 3.20.2
-- The C compiler identification is GNU 11.2.0
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /mnt/d/git/plutosdr-fw/buildroot/output/host/bin/arm-linux-gnueabihf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Performing Test HAS_WSHADOW
-- Performing Test HAS_WSHADOW - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strndup
-- Looking for strndup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for strtok_r
-- Looking for strtok_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for in6addr_any
-- Looking for in6addr_any - found
-- Looking for libusb_get_version
-- Looking for libusb_get_version - found
-- Found Git: /usr/bin/git (found version "2.36.0")
CMake Error at CMakeLists.txt:379 (message):
  Unable to find libzstd dependency.

  If you want to disable ZSTD compression support, set WITH_ZSTD=OFF.

-- Performing Test WITH_NETWORK_EVENTFD
-- Performing Test WITH_NETWORK_EVENTFD - Success
-- Performing Test Iconv_IS_BUILT_IN
-- Performing Test Iconv_IS_BUILT_IN - Success
-- Found BISON: /mnt/d/git/plutosdr-fw/buildroot/output/host/bin/bison (found version "3.8.2")
-- Found FLEX: /mnt/d/git/plutosdr-fw/buildroot/output/host/bin/flex (found version "2.6.4")
-- Looking for pthread_setname_np
-- Looking for pthread_setname_np - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of struct usb_functionfs_descs_head_v2
-- Check size of struct usb_functionfs_descs_head_v2 - done
INSTALL TARGETS - target iio has PUBLIC_HEADER files but no PUBLIC_HEADER DESTINATION.
-- Features enabled : xml zstd network dns-sd avahi ipv6 local local-dmabuf local-mmap usb utils iiod iiod-serial iiod-aio iiod-usb man-utils
-- Features disabled: serial hwmon examples modules usb-dynamic network-dynamic iiod-systemd iiod-sysvinit iiod-upstart doc man python-bindings c#-bindings
-- Configuring incomplete, errors occurred!
See also "/mnt/d/git/libiio/build/CMakeFiles/CMakeOutput.log".
```

catkira, 2022-08-03:
Ah I think the problem is that
CMAKE_FIND_ROOT_PATH = /mnt/d/git/plutosdr-fw/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot
and the libzstd.so is not present in /mnt/d/git/plutosdr-fw/buildroot/output/host/arm-buildroot-linux-gnueabihf/sysroot/lib
I am going to do a clean rebuild of buildroot...

catkira, 2022-08-04:
Its working now, I add to add zstd as dependency for libiio in buildroot
https://github.com/catkira/buildroot/commit/d1c0130beef58096ff00343cfce5c8d4a6b228ec

closed 2022-08-04 by catkira