analogdevicesinc/libiio #870: iio_attr cannot write attribute

catkira, 2022-08-09:
I am trying to set the rf_port_select attribute of a pluto with iio_attr, but it does not work.

```
# iio_attr -c ad9361-phy voltage0 rf_port_select
B_BALANCED
A
# iio_attr -c ad9361-phy voltage0 rf_port_select A
error Invalid argument (22) while writing 'rf_port_select' with 'A'
A
```
I also tried
```
# iio_attr -c ad9361-phy voltage0 rf_port_select 'A'
error Invalid argument (22) while writing 'rf_port_select' with 'A'
A
```

tfcollins, 2022-08-09:
rf_port_select is not configurable on Pluto as the ports are not physically connected to anything. This port lock is set in devicetree: https://github.com/analogdevicesinc/linux/blob/master/arch/arm/boot/dts/zynq-pluto-sdr.dtsi#L249

-Travis

catkira, 2022-08-09:
Ok thanks, that makes sense. Pluto only has RX1a, RX2a, TX1a, TX2a as is visible here:
https://wiki.analog.com/_media/university/tools/pluto/hacking/pluto_c.png 

A more comprehensive error message would be nice, or the _available attribute should only show the locked value. Or is it possible the query the lock status of an attribute somehow?

closed 2022-08-10 by tfcollins