analogdevicesinc/libiio #871: Trouble connecting to Pluto on Apple silicon with python bindings

kingsquirrel152, 2022-08-11:
Greetings, 

I am trying to connect to a Pluto SDR via python bindings. Note this is on Apple M2 silicon, os x 12.4.  and I've manually installed all the things via instructions here: https://pysdr.org/content/pluto.html#installing-plutosdr-driver

In the simplest form I am doing something simple like:

```python
sdr = adi.ad9361(uri='ip:192.168.3.1')
sdr.sample_rate = int(sample_rate_hz)
```

But the call to set the sample_rate blows up with a key error for the channel that is being used in the library.  I can do a `iio_addr` dump from the CLI and I see sampling frequency and all the usual suspects. I can also see the contexts for the pluto. So I think iiolib itself is happy.  It appears as if something is amiss in the python bindings.

```python
---------------------------------------------------------------------------
KeyError                                  Traceback (most recent call last)
Input In [10], in <cell line: 5>()
      1 sdr._rxadc.set_kernel_buffers_count(4)
      3 sdr.rx_enabled_channels = [0, 1]
----> 5 sdr.sample_rate = int(sample_rate_hz)
      7 # Config Rx
      8 sdr.rx_lo = int(center_freq_hz)

File ~/Library/Python/3.8/lib/python/site-packages/adi/ad936x.py:188, in ad9364.sample_rate(self, rate)
    186     taps = 64
    187 # fmt: on
--> 188 current_rate = self._get_iio_attr("voltage0", "sampling_frequency", False)
    190 if self._get_iio_attr("out", "voltage_filter_fir_en", False):
    191     if current_rate <= (25000000 // 12):

File ~/Library/Python/3.8/lib/python/site-packages/adi/attribute.py:156, in attribute._get_iio_attr(self, channel_name, attr_name, output, _ctrl)
    153 def _get_iio_attr(self, channel_name, attr_name, output, _ctrl=None):
    154     """ Get channel attribute as number """
    155     return get_numbers(
--> 156         self._get_iio_attr_str(channel_name, attr_name, output, _ctrl)
    157     )

File ~/Library/Python/3.8/lib/python/site-packages/adi/attribute.py:151, in attribute._get_iio_attr_str(self, channel_name, attr_name, output, _ctrl)
    149 if not channel:
    150     raise Exception("No channel found with name: " + channel_name)
--> 151 return channel.attrs[attr_name].value

KeyError: 'sampling_frequency'
```

Thanks in advance.

tfcollins, 2022-08-11:
I would suggest adding print statements around here: https://github.com/analogdevicesinc/libiio/blob/master/bindings/python/iio.py#L833

As we don't have Apple Silicon machines we cannot debug this directly. My guess is there is a bug in one of those functions. Here is what I would use to driver that code:

```
import iio
ctx = iio.Context('ip:pluto.local")
for dev in ctx.devices:
    for chan in dev.channels:
        for attr in chan.attrs:
            print(f'{dev.name}: {chan.name}: {attr.name}')
```

pcercuei, 2022-10-20:
@kingsquirrel152 Is that still a problem?

kingsquirrel152, 2022-10-20:
I have not had a chance to check. Apologies. We've since moved on to a different SDR. Im happy to close this if you like, or can leave it open and try to debug when I get a chance if thats helpful.

closed 2022-10-20 by tfcollins