analogdevicesinc/libiio #876: Problems when installing libiio 0.x and 1.x at the same time

catkira, 2022-08-24:
When I have libiio 0.x and 1.x installed at the same time on my host system, some apps that were compiled with 0.x dont work anymore.

This is my situation when libiio 0.x was installed last
```
benja@DESKTOP:/mnt/d/git/libiio_0.x/build$ ls -ls /usr/lib/libiio*
  0 lrwxrwxrwx 1 root root     11 Aug 24 21:31 /usr/lib/libiio.so -> libiio.so.0
  0 lrwxrwxrwx 1 root root     14 Aug 24 21:31 /usr/lib/libiio.so.0 -> libiio.so.0.23
628 -rw-r--r-- 1 root root 642176 Aug  8 18:11 /usr/lib/libiio.so.0.23
  0 lrwxrwxrwx 1 root root     13 Aug  8 13:19 /usr/lib/libiio.so.1 -> libiio.so.1.0
768 -rw-r--r-- 1 root root 784288 Aug  9 23:13 /usr/lib/libiio.so.1.0
```

SigDigger works then.

This is the situation when libiio 1.x was installed last
```
  0 lrwxrwxrwx 1 root root     11 Aug 24 21:37 /usr/lib/libiio.so -> libiio.so.1
 32 -rw-r--r-- 1 root root  29608 Aug  9 23:13 /usr/lib/libiio.so.0
628 -rw-r--r-- 1 root root 642176 Aug  8 18:11 /usr/lib/libiio.so.0.23
  0 lrwxrwxrwx 1 root root     13 Aug  8 13:19 /usr/lib/libiio.so.1 -> libiio.so.1.0
768 -rw-r--r-- 1 root root 784288 Aug  9 23:13 /usr/lib/libiio.so.1.0
```

SigDigger is not working in this case
```
benja@DESKTOP:/mnt/d/git/SoapySDR/Apps$ SigDigger
ERROR: Unable to create Avahi DNS-SD client :Daemon not running
ERROR: Unable to scan ip context: Text file busy (26)
[INFO] Opening PlutoSDR #0 192.168.137.2...
SigDigger: symbol lookup error: /usr/local/lib/SoapySDR/modules0.8/libPlutoSDRSupport.so: undefined symbol: iio_channel_attr_write
```

catkira, 2022-08-24:
Ahhh the was a trap for young players - the compat layer is on by default and that also installs a .so.0 lib.
@pcercuei Can You turn off the LIBIIO_COMPAT option by default as long as its not fully implemented?

pcercuei, 2022-08-25:
It's almost fully implemented in the `pcercuei/new-blocks-api-compat` branch, if you want to try.

What's not yet working is non-blocking refill/push, and cyclic buffers.

catkira, 2022-08-25:
When I compile sdrangel with the compat layer I get this error when executing:
`sdrangel: symbol lookup error: /usr/local/lib/sdrangel/libplutosdrdevice.so: undefined symbol: iio_create_scan_context`

catkira, 2022-08-25:
When I try to tell FindPkgInfo to look for libiio<1 it does this
```
-- Checking for module 'libiio<1'
--   Found libiio, version 0.23
-- Found libiio: /usr/include, /usr/lib/x86_64-linux-gnu/libiio.so
```
which points to 1.0 
```
benja@DESKTOP:/mnt/d/git/sdrangel$ ls -la /usr/lib/x86_64-linux-gnu/libiio.so
lrwxrwxrwx 1 root root 11 Aug 25 16:45 /usr/lib/x86_64-linux-gnu/libiio.so -> libiio.so.1
```

catkira, 2022-08-25:
ok, just for trying it out, I change the symlink manually
```
benja@DESKTOP:/mnt/d/git/libiio/build$ sudo ln -sf /usr/lib/libiio.so.0 /usr/lib/libiio.so
benja@DESKTOP:/mnt/d/git/libiio/build$ ls -la /usr/lib/libiio*
lrwxrwxrwx 1 root root     20 Aug 25 17:19 /usr/lib/libiio.so -> /usr/lib/libiio.so.0
-rw-r--r-- 1 root root 159384 Aug 25 16:46 /usr/lib/libiio.so.0
lrwxrwxrwx 1 root root     13 Aug 25 17:19 /usr/lib/libiio.so.1 -> libiio.so.1.0
-rw-r--r-- 1 root root 775520 Aug 25 16:46 /usr/lib/libiio.so.1.0
benja@DESKTOP:/mnt/d/git/libiio/build$
```

pcercuei, 2022-08-25:
You need to either:
* compile sdrangel with the v0.23 iio.h header, and link with the compat library's libiio.so.0.
* compile sdrangel with libiio v0.23 (uninstall v1.0 if needed), then uninstall v0.23 and install v1.0 + compat library.

catkira, 2022-08-25:
yes, but it would be nice if CMake would choose the right version of libiio when both versions are installed. I think thats what the version syntax of PkgConfig is for, like "libiio<1"
But its not doing it somehow.

catkira, 2022-08-25:
Even when I have libiio.so pointing to libiio.so.0 CMake is still linking against libiio.so.1 when I compile sdrangel. So I have to completely remove everything like libiio.so.1 in order to test the compat layer. I think thats not how it should be.

catkira, 2022-08-25:
Ahh no, it was the old problem that libiio 0.x is installing to  /usr/lib/x86_64-linux-gnu and libiio 1.x to /usr/lib

Can You fix that?

catkira, 2022-08-25:
Ok, now when I say to PkgConfig that I want libiio<1 I get this message
```
-- Checking for module 'libiio<1'
--   Requested 'libiio < 1' but version of libiio is 1.0
```
It would be nice, if PkgConfig could provide the compat layer for apps that require libiio<1. But I dont know if its possible to define 2 versions in the libiio.pc file

catkira, 2022-08-25:
Ok, now I get this error
`sdrangel: symbol lookup error: /usr/local/lib/sdrangel/libplutosdrdevice.so: undefined symbol: iio_device_identify_filename`
seems to be missing in the compat lib

catkira, 2022-08-25:
When I try to run SoapySDRUtil with the compat layer (SoapySDRPluto was compiled with libiio 0.23) I get this error:
```
benja@DESKTOP:/mnt/d/git/SoapyPlutoSDR/build$ SoapySDRUtil --rate="100000000" --direction=RX --args="driver=plutosdr,hostname=192.168.137.2"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Opening PlutoSDR #0 192.168.137.2...
[INFO] Using format CS16.
[INFO] Auto setting Buffer Size: 1
[INFO] Set MTU Size: 1
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 100 Msps
Starting stream loop, press Ctrl+C to exit...
ERROR: Unable to create buffer: Device or resource busy (16)
[ERROR] Unable to create buffer!
Error in rate test: Unable to create buffer!
```
this might be related to the missing buffer functions that You mentioned.

pcercuei, 2022-08-26:
@catkira force-pushed the branch with support for `iio_device_identify_filename`.

catkira, 2022-08-26:
nice. Did You also have a look at this problem:
```
ERROR: Unable to create buffer: Device or resource busy (16)
[ERROR] Unable to create buffer!
```
?

pcercuei, 2022-08-26:
I force-pushed `pcercuei/new-blocks-api-compat`, SoapySDRUtil works now with the compat API.

It does something that I didn't really expect - call iio_buffer_start() before iio_buffer_refill(), which shouldn't actually be supported. However since there is code out there doing this, and it works with v0.24, I updated the compat code to support this pattern as well.

catkira, 2022-08-26:
Now I get this error with SoapySDRUtil:
```
benja@DESKTOP:/mnt/d/git/libiio_cross$ SoapySDRUtil --rate="100000000" --direction=RX --args="driver=plutosdr,hostname=192.168.137.2"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

start
[INFO] Opening PlutoSDR #0 192.168.137.2...
[INFO] Using format CS16.
[INFO] Auto setting Buffer Size: 524288
[INFO] Set MTU Size: 524288
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 100 Msps
Starting stream loop, press Ctrl+C to exit...
Segmentation fault
```

catkira, 2022-08-26:
angelsdr is still failing with the same error:
`sdrangel: symbol lookup error: /usr/local/lib/sdrangel/libplutosdrdevice.so: undefined symbol: iio_device_identify_filename`

`benja@DESKTOP:/mnt/d/git/libiio/build$ readelf -s /usr/lib/libiio.so.0 | grep iio_device_identify_filename` shows that the function is not exported

pcercuei, 2022-08-26:
Well, the function *is* exported: https://github.com/analogdevicesinc/libiio/blob/pcercuei/new-blocks-api-compat/compat.c#L805-L810

So double-check that you are using the latest code.

catkira, 2022-08-26:
ah yes I forgot to do git fetch first ^^
But now I see that is installing to the "wrong" aka different folder again ... aaaaaaaaaah

catkira, 2022-08-26:
In CMakeLists.txt I had to outcomment:
```
               file(TO_NATIVE_PATH "${CMAKE_INSTALL_FULL_LIBDIR}/libiio/" IIO_MODULES_DIR)
               string(REPLACE "\\" "\\\\" IIO_MODULES_DIR ${IIO_MODULES_DIR})
```

Now I get with sdrangel this error
```
[INFO] [UHD] linux; GNU C++ version 9.4.0; Boost_107100; UHD_
sdrangel: symbol lookup error: /usr/local/lib/sdrangel/libplutosdrdevice.so: undefined symbol: iio_device_attr_read
```

catkira, 2022-08-26:
SoapySDRUtil gives this error now
```
benja@DESKTOP:/mnt/d/git/libiio_cross$ SoapySDRUtil --rate="100000000" --direction=RX --args="driver=plutosdr,hostname=192.168.137.2"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

start
[INFO] Opening PlutoSDR #0 192.168.137.2...
SoapySDRUtil: symbol lookup error: /usr/local/lib/SoapySDR/modules0.8/libPlutoSDRSupport.so: undefined symbol: iio_channel_attr_write
```

catkira, 2022-08-26:
ah no i was in the new-block-api branch, too late already^^

catkira, 2022-08-26:
SoapySDRUtil is working now. 
This is with 0.24 on the device:
benja@DESKTOP-NSB1HR8:/mnt/d/git/libiio_cross$ SoapySDRUtil --rate="100000000" --direction=RX --args="driver=plutosdr,hostname=192.168.137.2"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

start
[INFO] Opening PlutoSDR #0 192.168.137.2...
[INFO] Using format CS16.
[INFO] Auto setting Buffer Size: 524288
[INFO] Set MTU Size: 524288
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 100 Msps
Starting stream loop, press Ctrl+C to exit...
[INFO] Has direct RX copy: 1
13.4824 Msps    53.9297 MBps
13.9468 Msps    55.7872 MBps
14.1011 Msps    56.4046 MBps

catkira, 2022-08-26:
With 1.0 running on the device I get this error:
```
benja@DESKTOP:/mnt/d/git/libiio_cross$ SoapySDRUtil --rate="100000000" --direction=RX --args="driver=plutosdr,hostname=192.168.137.2"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

start
[INFO] Opening PlutoSDR #0 192.168.137.2...
[INFO] Using format CS16.
ERROR: Unable to dequeue block: Connection timed out (110)
[ERROR] Unable to create buffer!
Error in rate test: Unable to create buffer!
```

pcercuei, 2022-08-26:
Use #877 on the device side.

catkira, 2022-08-26:
So the compat layer seems to work better when 0.x is running on the device. In that case I can listen to FM radio with SigDigger (using the Soapy plugin). With sdrangel I get only garbage signal (sdrangel uses libiio directly).

catkira, 2022-08-26:
Soapy still gives this error:
```
benja@DESKTOP:/mnt/d/git/libiio_cross$ SoapySDRUtil --rate="100000000" --direction=RX --args="driver=plutosdr,hostname=192.168.137.2"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

start
[INFO] Opening PlutoSDR #0 192.168.137.2...
[INFO] Using format CS16.
ERROR: Unable to dequeue block: Connection timed out (110)
[ERROR] Unable to create buffer!
Error in rate test: Unable to create buffer!
```

catkira, 2022-08-26:
sdrangel does this:
```
ERROR: Unable to enqueue block: Broken pipe (32)
2022-08-27 00:04:05.950 (W) PlutoSDRInputThread::run: error refilling buf -32 / 65536
ERROR: Unable to enqueue block: Broken pipe (32)
2022-08-27 00:04:06.150 (W) PlutoSDRInputThread::run: error refilling buf -32 / 65536
ERROR: Unable to enqueue block: Broken pipe (32)
2022-08-27 00:04:06.351 (W) PlutoSDRInputThread::run: error refilling buf -32 / 65536
ERROR: Unable to enqueue block: Broken pipe (32)
2022-08-27 00:04:06.551 (W) PlutoSDRInputThread::run: error refilling buf -32 / 65536
ERROR: Unable to enqueue block: Broken pipe (32)
2022-08-27 00:04:06.751 (W) PlutoSDRInputThread::run: error refilling buf -32 / 65536
ERROR: Unable to enqueue block: Broken pipe (32)
```

catkira, 2022-08-26:
and SigDigger this
[image]

catkira, 2022-08-26:
iio_rwdev is also not working anymore
```
benja@DESKTOP:/mnt/d/git/libiio_cross$ iio_rwdev -u ip:192.168.137.2 -B -b 65768 cf-ad9361-lpc
ERROR: Unable to create IIO context ip:192.168.137.2: Connection refused (111)
```

catkira, 2022-08-28:
or was I supposed to use the fixes branch on device _and_ host?
I only used it on device ...

pcercuei, 2022-08-29:
@catkira the compat library is just a few days old, and the new API and protocol are still pretty new, so you are pretty much in uncharted territory ;)

What I can already tell is that the "broken pipe" and "connection refused" errors occur when IIOD crashed on the target.

catkira, 2022-09-06:
I think this is already solved. It was probably caused by a combination of https://github.com/analogdevicesinc/libiio/issues/889 and https://github.com/analogdevicesinc/libiio/issues/890 .

closed 2022-09-06 by catkira