analogdevicesinc/libiio #888: Signal buffer overflows

catkira, 2022-08-29:
Add a feature to signal buffer overflows

Motivation:
If the sample rate of the device is too fast for the device CPU or network connection, the buffer in iiod overflows and samples get lost. It would be good if the user knows when that happens

Implementation:
It could be enabled via a parameter to iiod. iiod could then write a message in case of overflow that is visible in dmesg.

pcercuei, 2022-09-08:
This is kind of a duplicate to #860.

IIOD does not know about overflows or underflows if the layers below don't report them. Having a sample counter mechanism (for RX) and a timestamp API (for TX) would allow reporting these underflows and overflows to the kernel, libiio, and eventually the applications.

catkira, 2022-09-08:
IIOD does not currently know about overflows in the hdl code, but it could already know overflows that occur when the network bandwidth cannot keep up with the samples coming from the device. Something like measuring the time it takes to write the data to the network socket and if that takes longer than the time reading the samples from device, then its clear that the network bandwidth is not enough and overflows occur. But I agree it would be a bit hacky, doing it with the timestamp functionality would be much cleaner.
Lets see if some people want to join developing the timestamp feature in hdl, if nothing else works I will do it myself. In fact I am waiting quite some time already, since libiio 1.0 is in a working state now and ready to go.