analogdevicesinc/libiio #889: lib installation dir is different for libiio 0.24 and 1.0

catkira, 2022-08-29:
Currently libiio 0.24 installs on my system (ubuntu 20) into `/usr/lib/x86_64-linux-gnu/` .
libiio 1.0 installs on the same system into `/usr/lib`. This causes problems.

@pcercuei can You make both versions install into the same dir?

As a quick fix I outcomment `include(GNUInstallDirs)` in CMakeLists.txt of libiio 0.24. This makes it also install to `/usr/lib` .

pcercuei, 2022-08-29:
Both install to `/usr/lib/x86_64-linux-gnu/` here (as they should).

Please try with a newer version of CMake. If it fixes the problem, please report what version of CMake you were using before.

catkira, 2022-08-29:
cmake version 3.16.3

pcercuei, 2022-09-01:
So did a newer version fix the problem then?

catkira, 2022-09-01:
Thats the version in the ubuntu 20 system. If I want a newer I would have to use a ppa. But there should be a Cmake requirement in in CMakeLists.txt .

pcercuei, 2022-09-01:
It's not about that. I need to know what I have to do to reproduce the problem and fix it.

catkira, 2022-09-01:
I just noticed that 
the  `origin/pcercuei/new-blocks-api-compat` branch installs to `/usr/lib`
and
the  `origin/pcercuei/dev-new-iface-fixes` branch installs to `/usr/lib/x86_64-linux-gnu/`

tested with Cmake  3.16.3

catkira, 2022-09-01:
Its very strange. I added a debug output in CMakeLists.txt

```
include(GNUInstallDirs)
message(STATUS ${CMAKE_INSTALL_FULL_LIBDIR})
```

the same line gives different outputs for libiio 0.24 and 1.0

```
benja@LAPTOP:/mnt/d/git/libiio_0.x/build$ cmake ..
-- cmake version: 3.16.3
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- /usr/lib/x86_64-linux-gnu
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Performing Test HAS_WSHADOW
^C
benja@LAPTOP:/mnt/d/git/libiio_0.x/build$ cd ..
benja@LAPTOP:/mnt/d/git/libiio_0.x$ cd ..
benja@LAPTOP:/mnt/d/git$ cd libiio
libiio/       libiio_0.x/   libiio_cross/
benja@LAPTOP-Benjamin:/mnt/d/git$ cd libiio
benja@LAPTOP-Benjamin:/mnt/d/git/libiio$ cd build
benja@LAPTOP-Benjamin:/mnt/d/git/libiio/build$ rm -rf *
benja@LAPTOP-Benjamin:/mnt/d/git/libiio/build$ cmake ..
-- cmake version: 3.16.3
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- /usr/local/lib
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Performing Test HAS_WSHADOW
```

catkira, 2022-09-01:
I think I found the problem:

```
include(GNUInstallDirs)
message(STATUS ${CMAKE_INSTALL_FULL_LIBDIR})
# Set the default install path to /usr
if (NOT APPLE AND NOT WIN32 AND CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT)
        set(CMAKE_INSTALL_PREFIX "/usr" CACHE PATH "default install path" FORCE)
endif()
include(GNUInstallDirs)
message(STATUS ${CMAKE_INSTALL_FULL_LIBDIR})
```

generates this output
```
-- /usr/local/lib
-- /usr/lib/x86_64-linux-gnu
```

This part is missing in libiio 1.0
```
if (NOT APPLE AND NOT WIN32 AND CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT)
        set(CMAKE_INSTALL_PREFIX "/usr" CACHE PATH "default install path" FORCE)
endif()
```

catkira, 2022-09-01:
This is the fix for origin/pcercuei/new-blocks-api-compat:
```
diff --git a/CMakeLists.txt b/CMakeLists.txt
index 6ca426fa..e7227499 100644
--- a/CMakeLists.txt
+++ b/CMakeLists.txt
@@ -16,6 +16,10 @@ if (WIN32)
        string(REPLACE "." "," LIBIIO_FILEVERSION ${VERSION})
 endif()

+if (NOT APPLE AND NOT WIN32 AND CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT)
+        set(CMAKE_INSTALL_PREFIX "/usr" CACHE PATH "default install path" FORCE)
+endif()
+
 if (NOT CMAKE_BUILD_TYPE)
        set(CMAKE_BUILD_TYPE RelWithDebInfo CACHE STRING
                "Choose the type of build, options are: None(CMAKE_CXX_FLAGS or CMAKE_C_FLAGS used) Debug Release RelWithDebInfo MinSizeRel."
```

pcercuei, 2022-09-01:
It is not missing, it is moved to `cmake/Install.cmake`.

I still can't reproduce it on the new-blocks-api-compat branch :(

pcercuei, 2022-09-01:
Ok, it does happen with CMake 3.18, and does not happen with CMake 3.24.

catkira, 2022-09-01:
In cmake/Install.cmake I added this
```
# Set the default install path to /usr
if (NOT WIN32 AND CMAKE_INSTALL_PREFIX_INITIALIZED_TO_DEFAULT)
        set(CMAKE_INSTALL_PREFIX "/usr" CACHE PATH "default install path" FORCE)
endif()
message(STATUS 1${CMAKE_INSTALL_FULL_LIBDIR})
include(GNUInstallDirs)
message(STATUS 2${CMAKE_INSTALL_FULL_LIBDIR})
```

```
benja@LAPTOP:/mnt/d/git/libiio/build$ cmake ..
-- cmake version: 3.16.3
-- The C compiler identification is GNU 9.4.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Performing Test HAS_WPEDANTIC
-- Performing Test HAS_WPEDANTIC - Success
-- Performing Test HAS_WSHADOW
-- Performing Test HAS_WSHADOW - Success
-- Looking for strdup
-- Looking for strdup - found
-- Looking for strndup
-- Looking for strndup - found
-- Looking for strerror_r
-- Looking for strerror_r - found
-- Looking for strtok_r
-- Looking for strtok_r - found
-- Looking for newlocale
-- Looking for newlocale - found
-- Looking for in6addr_any
-- Looking for in6addr_any - found
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29.1")
-- Checking for module 'libusb-1.0'
--   Found libusb-1.0, version 1.0.23
-- Looking for libusb_get_version
-- Looking for libusb_get_version - found
-- Found Git: /usr/bin/git (found version "2.25.1")
-- Performing Test WITH_NETWORK_EVENTFD
-- Performing Test WITH_NETWORK_EVENTFD - Success
-- Found LibXml2: /usr/lib/x86_64-linux-gnu/libxml2.so (found version "2.9.10")
-- Found BISON: /usr/bin/bison (found version "3.5.1")
-- Found FLEX: /usr/bin/flex (found version "2.6.4")
-- Looking for pthread_setname_np
-- Looking for pthread_setname_np - found
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of struct usb_functionfs_descs_head_v2
-- Check size of struct usb_functionfs_descs_head_v2 - done
-- 1/usr/local/lib
-- 2/usr/lib/x86_64-linux-gnu
-- Features enabled : xml zstd network dns-sd avahi ipv6 local local-dmabuf local-mmap usb utils iiod iiod-serial iiod-aio iiod-usb man-utils
-- Features disabled: serial hwmon examples modules usb-dynamic network-dynamic iiod-systemd iiod-sysvinit iiod-upstart doc man python-bindings c#-bindings
-- Configuring done
-- Generating done
-- Build files have been written to: /mnt/d/git/libiio/build
```

catkira, 2022-09-01:
I think its better to make it work with cmake 3.16 than requiring 3.24, because Ubuntu 20 is still widely used.

pcercuei, 2022-09-01:
Yes of course, we actually want to keep compatibility with Ubuntu 18.04 which uses CMake 3.10.

I created a PR with a fix and a short description of the problem.

catkira, 2022-09-01:
ok. Can You rebase the fix and compat branch on to the latest dev branch after the PR has been merged (otherwise I have to manually patch it many times when testing out new stuff)?

pcercuei, 2022-09-01:
Sure. Then I think you can use the compat branch on both sides (host and target) and it should work fine.

catkira, 2022-09-01:
Nice. In the meantime You can try SigDigger with the compat layer ;)

pcercuei, 2022-09-08:
Fixed in the dev branch.

PR #892 (2022-09-08): CMake: Fix libraries not installed to the right path on older CMake
On CMake 3.18, the libraries were installed to /usr/lib, meanwhile on
CMake 3.24 they were installed to /usr/lib/<target_triple>, where they
should be.

This seemed to happen because CMAKE_INSTALL_PREFIX was set after the
include(GNUInstallDirs).

Work around this issue by making sure that CMAKE_INSTALL_PREFIX is set
as early as possible.

Fixes #889.
commit 4820ec49 (2022-09-08): CMake: Fix libraries not installed to the right path on older CMake

On CMake 3.18, the libraries were installed to /usr/lib, meanwhile on
CMake 3.24 they were installed to /usr/lib/<target_triple>, where they
should be.

This seemed to happen because CMAKE_INSTALL_PREFIX was set after the
include(GNUInstallDirs).

Work around this issue by making sure that CMAKE_INSTALL_PREFIX is set
as early as possible.

Fixes #889.
closed 2022-09-08 by pcercuei