analogdevicesinc/libiio #890: Changing sample rate with the compat layer does not work

catkira, 2022-08-29:
When using SigDigger with libiio 1.0 compat layer, changing the sample rate does not work. Values in the GUI are ignored. When I start the stream it uses the sample rate that is currently active on the device.

ie when I do `echo 60000000 > /sys/bus/iio/devices/iio:device0/in_voltage_sampling_frequency` on the device, SigDigger shows 60 MSPS even though I chose 1 MSPS in the GUI.

catkira, 2022-08-29:
changing the RF frequency also does not seem to work

catkira, 2022-08-29:
I think its a good test if You compile SigDigger and try to listen to a local FM station with the FM decoder.

catkira, 2022-08-30:
Its a bit tricky to get SigDigger to work with pluto (or any other libiio device) if avahi is not working (which is the case if You use WSL). You have to add these device specific parameters
[image]

pcercuei, 2022-09-01:
Please wait until the compat layer actually makes it to the dev branch, at least.

pcercuei, 2022-09-01:
Are you talking about the old compat.c code that's already in the dev branch, or the latest code in the `pcercuei/new-blocks-api-compat` branch?

catkira, 2022-09-01:
pcercuei/new-blocks-api-compat

pcercuei, 2022-09-01:
Make sure your IIOD is built using the branch from this PR: https://github.com/analogdevicesinc/libiio/pull/877
I fixed a bug where IIOD would ignore "write attribute" commands.

catkira, 2022-09-01:
with [this](https://github.com/analogdevicesinc/libiio/pull/877) I cannot compile SoapyPlutoSDR.
I changed libiio.o to point to the compat lib with this command:
 `sudo ln -sf /usr/lib/x86_64-linux-gnu/libiio.so.0 /usr/lib/x86_64-linux-gnu/libiio.so`

Our should I use the compat branch for the host? Its a bit confusing with so many different branches ^^

```
benja@LAPTOP:/mnt/d/git/SoapyPlutoSDR/build$ make
[ 20%] Building CXX object CMakeFiles/PlutoSDRSupport.dir/PlutoSDR_Registration.cpp.o
In file included from /mnt/d/git/SoapyPlutoSDR/PlutoSDR_Registration.cpp:7:
/usr/include/libusb-1.0/libusb.h:67:26: warning: ISO C++ forbids zero-size array ‘dev_capability_data’ [-Wpedantic]
   67 | #define ZERO_SIZED_ARRAY 0 /* [0] - non-standard, but usually working code */
      |                          ^
/usr/include/libusb-1.0/libusb.h:741:30: note: in expansion of macro ‘ZERO_SIZED_ARRAY’
  741 |  uint8_t dev_capability_data[ZERO_SIZED_ARRAY];
      |                              ^~~~~~~~~~~~~~~~
/usr/include/libusb-1.0/libusb.h:67:26: warning: ISO C++ forbids zero-size array ‘dev_capability’ [-Wpedantic]
   67 | #define ZERO_SIZED_ARRAY 0 /* [0] - non-standard, but usually working code */
      |                          ^
/usr/include/libusb-1.0/libusb.h:766:62: note: in expansion of macro ‘ZERO_SIZED_ARRAY’
  766 |  struct libusb_bos_dev_capability_descriptor *dev_capability[ZERO_SIZED_ARRAY];
      |                                                              ^~~~~~~~~~~~~~~~
/usr/include/libusb-1.0/libusb.h:67:26: warning: ISO C++ forbids zero-size array ‘iso_packet_desc’ [-Wpedantic]
   67 | #define ZERO_SIZED_ARRAY 0 /* [0] - non-standard, but usually working code */
      |                          ^
/usr/include/libusb-1.0/libusb.h:1259:54: note: in expansion of macro ‘ZERO_SIZED_ARRAY’
 1259 |  struct libusb_iso_packet_descriptor iso_packet_desc[ZERO_SIZED_ARRAY];
      |                                                      ^~~~~~~~~~~~~~~~
/mnt/d/git/SoapyPlutoSDR/PlutoSDR_Registration.cpp: In function ‘std::vector<std::map<std::__cxx11::basic_string<char>, std::__cxx11::basic_string<char> > > find_PlutoSDR(const Kwargs&)’:
/mnt/d/git/SoapyPlutoSDR/PlutoSDR_Registration.cpp:18:2: error: ‘iio_scan_context’ was not declared in this scope; did you mean ‘iio_create_context’?
   18 |  iio_scan_context *scan_ctx;
      |  ^~~~~~~~~~~~~~~~
      |  iio_create_context
```

catkira, 2022-09-01:
I managed to compile SoapyPlutoSDR by installing libiio 0.24 again and after compiling installing the compat branch again.
SigDigger works partly. When switching rx on the first time data is received. But after changing the sample rate this error pops up
[image]

pcercuei, 2022-09-01:
@catkira No idea how to use that program. And it doesn't detect my Pluto?

catkira, 2022-09-01:
If you have SoapySDR and SoapyPlutoSDR installed, make sure it works by testing this:
` SoapySDRUtil --rate="100000000" --direction=RX --args="driver=plutosdr,hostname=192.168.137.2"`

After that add these parameters in the SigDigger GUI:
[image]
Then after restarting SigDigger, Your Pluto should get detected. If not, let me know again.

pcercuei, 2022-09-01:
Testing with CubicSDR + SoapyPlutoSDR and the compat layer, I can change the sample rate just fine and I have no problem starting / stopping the stream.

catkira, 2022-09-01:
The compat layer also works with SigDigger when I run 0.24 on the device.

catkira, 2022-09-01:
Trying to compile CubicSDR....
```
benja@LAPTOP:/mnt/d/git/CubicSDR/build$ make
Scanning dependencies of target CubicSDR
[  1%] Building CXX object CMakeFiles/CubicSDR.dir/external/rtaudio/RtAudio.cpp.o
[  2%] Building CXX object CMakeFiles/CubicSDR.dir/src/CubicSDR.cpp.o
In file included from /mnt/d/git/CubicSDR/src/demod/DemodulatorInstance.h:11,
                 from /mnt/d/git/CubicSDR/src/demod/DemodulatorMgr.h:10,
                 from /mnt/d/git/CubicSDR/src/visual/PrimaryGLContext.h:14,
                 from /mnt/d/git/CubicSDR/src/CubicSDR.h:13,
                 from /mnt/d/git/CubicSDR/src/CubicSDR.cpp:17:
/mnt/d/git/CubicSDR/src/modules/modem/ModemDigital.h:45:53: error: ‘modemcf’ has not been declared
   45 |     virtual void digitalStart(ModemKitDigital *kit, modemcf mod, ModemIQData *input);
```

pcercuei, 2022-09-01:
I just apt-get install'd it, and it worked :)
Note that it doesn't work over USB yet, I'm working on it.

catkira, 2022-09-01:
yes, but then it installs so many dependcy libs that I have already installed manually in usr/local, therefore I want to avoid that

catkira, 2022-09-01:
cant get CubicSDR to compile, I am going to skip this one ^^

catkira, 2022-09-02:
I ported SoapyPlutoSDR quickly to libiio 1.0 (see [here](https://github.com/catkira/SoapyPlutoSDR/tree/libiio_1.0)). However changing frequency or sample rate does not work.
Changing the frequency is done like this:
```c
iio_channel_attr_write_longlong(iio_device_find_channel(dev, "altvoltage0", true),"frequency", freq);
```

catkira, 2022-09-02:
yes, when I change the sample rate in ad9361 by doing this
```diff
diff --git a/examples/ad9361-iiostream.c b/examples/ad9361-iiostream.c
index ad8af902..9602cf35 100644
--- a/examples/ad9361-iiostream.c
+++ b/examples/ad9361-iiostream.c
@@ -204,13 +204,13 @@ int main (int argc, char **argv)

        // RX stream config
        rxcfg.bw_hz = MHZ(2);   // 2 MHz rf bandwidth
-       rxcfg.fs_hz = MHZ(2.5);   // 2.5 MS/s rx sample rate
+       rxcfg.fs_hz = MHZ(20.5);   // 2.5 MS/s rx sample rate
        rxcfg.lo_hz = GHZ(2.5); // 2.5 GHz rf frequency
        rxcfg.rfport = "A_BALANCED"; // port A (select for rf freq.)

        // TX stream config
        txcfg.bw_hz = MHZ(1.5); // 1.5 MHz rf bandwidth
-       txcfg.fs_hz = MHZ(2.5);   // 2.5 MS/s tx sample rate
+       txcfg.fs_hz = MHZ(20.5);   // 2.5 MS/s tx sample rate
        txcfg.lo_hz = GHZ(2.5); // 2.5 GHz rf frequency
        txcfg.rfport = "A"; // port A (select for rf freq.)
```

it does not seem to have an effect. The output is still the same:
```
* Starting IO streaming (press CTRL+C to cancel)
        RX     0.26 MSmp, TX     0.26 MSmp
        RX     0.52 MSmp, TX     0.52 MSmp
        RX     0.79 MSmp, TX     0.79 MSmp
        RX     1.05 MSmp, TX     1.05 MSmp
        RX     1.31 MSmp, TX     1.31 MSmp
        RX     1.57 MSmp, TX     1.57 MSmp
```

catkira, 2022-09-02:
I tried to read back the sample rate like this, but it does not work:
```diff
diff --git a/examples/ad9361-iiostream.c b/examples/ad9361-iiostream.c
index ad8af902..3a72fc2b 100644
--- a/examples/ad9361-iiostream.c
+++ b/examples/ad9361-iiostream.c
@@ -162,9 +162,13 @@ bool cfg_ad9361_streaming_ch(struct stream_cfg *cfg, enum iodev type, int chid)
        // Configure phy and lo channels
@@ -162,9 +162,13 @@ bool cfg_ad9361_streaming_ch(struct stream_cfg *cfg, enum iodev type, int chid)
@@ -162,9 +162,13 @@ bool cfg_ad9361_streaming_ch(struct stream_cfg *cfg, enum iodev type, int chid)
        // Configure phy and lo channels
        printf("* Acquiring AD9361 phy channel %d\n", chid);
        if (!get_phy_chan(type, chid, &chn)) {  return false; }
-       wr_ch_str(chn, "rf_port_select",     cfg->rfport);
+       //wr_ch_str(chn, "rf_port_select",     cfg->rfport);
        wr_ch_lli(chn, "rf_bandwidth",       cfg->bw_hz);
+       printf("write sampling frequency %lu\n", cfg->fs_hz);
        wr_ch_lli(chn, "sampling_frequency", cfg->fs_hz);
+       long long val = 0;
+       iio_channel_attr_read_longlong(chn, "sampling_frequency", &val);
+       printf("read sampling frequency = %lu\n", val);

        // Configure LO channel
        printf("* Acquiring AD9361 %s lo channel\n", type == TX ? "TX" : "RX");
@@ -204,13 +208,13 @@ int main (int argc, char **argv)

        // RX stream config
        rxcfg.bw_hz = MHZ(2);   // 2 MHz rf bandwidth
-       rxcfg.fs_hz = MHZ(2.5);   // 2.5 MS/s rx sample rate
+       rxcfg.fs_hz = MHZ(20.5);   // 2.5 MS/s rx sample rate
        rxcfg.lo_hz = GHZ(2.5); // 2.5 GHz rf frequency
        rxcfg.rfport = "A_BALANCED"; // port A (select for rf freq.)

        // TX stream config
        txcfg.bw_hz = MHZ(1.5); // 1.5 MHz rf bandwidth
-       txcfg.fs_hz = MHZ(2.5);   // 2.5 MS/s tx sample rate
+       txcfg.fs_hz = MHZ(20.5);   // 2.5 MS/s tx sample rate
        txcfg.lo_hz = GHZ(2.5); // 2.5 GHz rf frequency
        txcfg.rfport = "A"; // port A (select for rf freq.)
```
This is the output when I run it
```
# ad9361-iiostream
* Acquiring IIO context
* Acquiring AD9361 streaming devices
* Configuring AD9361 for streaming
* Acquiring AD9361 phy channel 0
write sampling frequency 1205
read sampling frequency = 0
* Acquiring AD9361 RX lo channel
* Acquiring AD9361 phy channel 0
write sampling frequency 1205
read sampling frequency = 0
* Acquiring AD9361 TX lo channel
* Initializing AD9361 IIO streaming channels
* Enabling IIO streaming channels
* Creating non-cyclic IIO buffers with 1 MiS
* Starting IO streaming (press CTRL+C to cancel)
        RX     0.26 MSmp, TX     0.26 MSmp
        RX     0.52 MSmp, TX     0.52 MSmp
        RX     0.79 MSmp, TX     0.79 MSmp
```

catkira, 2022-09-02:
When I run the same programm on host (`with wr_ch_lli(chn, "rf_bandwidth",cfg->bw_hz);` out commented because of https://github.com/analogdevicesinc/libiio/issues/895), I get this error:
```
benja@LAPTOP:/mnt/d/git/libiio/build/examples$ ./ad9361-iiostream ip:192.168.137.2
* Acquiring IIO context
* Acquiring AD9361 streaming devices
* Configuring AD9361 for streaming
* Acquiring AD9361 phy channel 0
write sampling frequency 2500000
Error -22 writing to channel "sampling_frequency"
value may not be supported.
* Destroying streams
* Destroying buffers
* Destroying channel masks
* Destroying context
```

catkira, 2022-09-02:
@pcercuei do You have an idea whats wrong here?

catkira, 2022-09-02:
@pcercuei 
I tried again using this https://github.com/analogdevicesinc/libiio/commit/2b78dd5a1c1c02c6dd6256bd80f91d00f5953c55 on host and device. The same problem.

catkira, 2022-09-02:
Something is weird. This function gets called with op code 10 (IIOD_OP_WRITE_CHN_ATTR), but the function cannot handle it.
```c
static ssize_t attr_write(struct parser_pdata *pdata,
			  const struct iiod_command *cmd,
			  const char *attr, const void *buf, size_t len)
{
	const struct iio_channel *chn;
	const struct iio_device *dev;
	uint16_t arg2 = cmd->code & 0xffff;

	dev = iio_context_get_device(pdata->ctx, cmd->dev);

	printf("attr_write cmd->op = %d cmd->code = %d attr = %s\n", cmd->op, cmd->code, attr);

	switch (cmd->op) {
	case IIOD_OP_READ_ATTR:
		return iio_device_attr_write_raw(dev, attr, buf, len);
	case IIOD_OP_READ_DBG_ATTR:
		return iio_device_debug_attr_write_raw(dev, attr, buf, len);
	case IIOD_OP_READ_BUF_ATTR:
		return iio_device_buffer_attr_write_raw(dev, arg2,
							attr, buf, len);
	case IIOD_OP_READ_CHN_ATTR:
		chn = iio_device_get_channel(dev, arg2);
		return iio_channel_attr_write_raw(chn, attr, buf, len);
	default:
		return -EINVAL;
	}
}
```

catkira, 2022-09-02:
I have created a PR that fixes this: https://github.com/analogdevicesinc/libiio/pull/897

PR #899 (2022-09-03): iiod: Fix writing of attributes
There was a copy paste error, all the opcodes were xxx_READ_xxx,
but they had to be xxx_WRITE_xxx.

Fixes: #890
commit 2bde8dad (2022-09-03): iiod: Fix writing of attributes

There was a copy paste error, all the opcodes were xxx_READ_xxx,
but they had to be xxx_WRITE_xxx.

Fixes: #890
closed 2022-09-03 by catkira