analogdevicesinc/libiio #891: lost kernel checks

rgetz, 2022-08-29:
during a shuffling of the CI directories, CI/travis/check_kernel.sh was removed.

https://github.com/analogdevicesinc/libiio/commit/09c08985ad1205f8ec4e40247e0737ef3b089f5d#diff-a37e1affdcbdf72de04029402805155c7cafde708598281cda8a923c0607e461

now, there is no checking to make sure the libiio enum and struct matches the kernel, which will eventually lead to some problems...

Add the file back:
https://github.com/analogdevicesinc/libiio/blob/5b9ef36ebccfc82533c5dd949b19f8180a751952/CI/travis/check_kernel.sh

and plumb it into the CI, so it works and checks things once per PR. (not on every CI client).

-Robin

ccraluca, 2022-09-01:
So this script should run on a CI machine for each PR?

In that commit are there other scripts that should not have been deleted?

-Raluca

rgetz, 2022-09-01:
> So this script should run on a CI machine for each PR?

yes - that is the way it use to work - the CI would make sure the libiio was sync'ed with the kernel on a each PR event - there wasn't any way at the time to do it once a week in a cron or something (and have tests fail) - so it was put into the CI system. If the kernel updates - everyone's PRs fail until it is fixed. 

Makes sure that everything is fixed quickly.

>In that commit are there other scripts that should not have been deleted?

I don't think so - as long as the other things are working...

commit 9f7894ca (2022-09-06): CI:add kernel check

It checks that the structures defined in iio.h are up-to-date with the
one defined in the kernel's user-space API headers.

Fixes: #891
PR #896 (2022-09-06): CI:add kernel check
Add kernel check. It checks that the structures defined in iio.h are up-to-date with the one defined in the kernel's user-space API headers.
Script will run for each PR, inside Linux-Ubuntu-22.04 job.

Fixes: #891
commit 5b46fa13 (2022-09-06): CI:add kernel check

It checks that the structures defined in iio.h are up-to-date with the
one defined in the kernel's user-space API headers.

Fixes: #891
closed 2022-09-06 by pcercuei