analogdevicesinc/libiio #898: libiio 1.0 does not like certain buffer sizes

catkira, 2022-09-02:
When I chose buffer size 100000 everything is fine:
[image]

But when I chose buffer size there is sporadic garbage data:
[image]

Its tested with libiio https://github.com/catkira/libiio/commit/76b2257ef2e2591a907bd0d5f60db6cfb3ee6258running on host and device.

catkira, 2022-09-02:
I observed that the pattern of garbage data changes when I change the number of blocks used. When I increase the number of blocks, the garbage data lines are spread further apart.

catkira, 2022-09-02:
And it also depends on the sample rate. Buffer size 100000 is ok with 10 or 5 MSPS but not ok with 2 or 1 MSPS.

catkira, 2022-09-02:
never mind, its working now, I was not using direct copy ^^ 
The samples are already interleaved with libiio 1.0.
@pcercuei is this always the case? Or is it possible that the first half of a block is I and the second half of a block Q samples?

pcercuei, 2022-09-02:
It's always the case, that's how the hardware delivers the samples.

catkira, 2022-09-03:
ok thanks.

closed 2022-09-03 by catkira