analogdevicesinc/libiio #901: ad9361-iiostream crashes when running on host

catkira, 2022-09-03:
ad9361 crashes when running on host (tested with the latest dev branch):
```
benja@LAPTOP:/mnt/d/git/libiio/build/examples$ ./ad9361-iiostream ip:192.168.137.2
* Acquiring IIO context
* Acquiring AD9361 streaming devices
* Configuring AD9361 for streaming
* Acquiring AD9361 phy channel 0
Error A_BALANCED writing to channel "rf_port_select"
value may not be supported. This error can be ignored for PlutoSDR and derivatives.
* Acquiring AD9361 RX lo channel
* Acquiring AD9361 phy channel 0
Error A writing to channel "rf_port_select"
value may not be supported. This error can be ignored for PlutoSDR and derivatives.
* Acquiring AD9361 TX lo channel
* Initializing AD9361 IIO streaming channels
* Enabling IIO streaming channels
* Creating non-cyclic IIO buffers with 1 MiS
* Starting IO streaming (press CTRL+C to cancel)
        RX     0.26 MSmp, TX     0.26 MSmp
ERROR: Unable to dequeue block: Bad file descriptor (9)
ERROR: Unable to send block: Bad file descriptor (9)
* Destroying streams
* Destroying buffers
* Destroying channel masks
* Destroying context
```
it is working when I run in directly on device and it is also working on host when I outcomment all tx stuff.

pcercuei, 2022-09-06:
Works fine here. With firmware v0.34 on the Pluto.

catkira, 2022-09-06:
I experienced the crash when using libiio 1.0 on host _and_ device.

commit d6ba7c02 (2022-09-08): stream: Avoid dequeueing already dequeued buffers

Dequeueing already dequeued buffers is now an error. Update the
iio_stream code to avoid doing this.

Fixes: #901.
commit e2b3d6d2 (2022-09-09): stream: Avoid dequeueing already dequeued buffers

Dequeueing already dequeued buffers is now an error. Update the
iio_stream code to avoid doing this.

Fixes: #901.
PR #903 (2022-09-12): iiod-client: Dequeueing an already dequeued block is not an error
Dequeueing an already dequeued block is not an error, but a no-op.
Return 0 instead of -EBADF.

Fixes: #901.
commit 3dcb1f94 (2022-09-12): stream: Avoid dequeueing already dequeued buffers

Dequeueing already dequeued buffers is now an error. Update the
iio_stream code to avoid doing this.

Fixes: #901.
closed 2022-09-12 by catkira