analogdevicesinc/libiio #909: Data reading from an STM32 pin with Python.

andresbove, 2022-09-30:
Hi! I'm working with an STM32MP157F-DK2, it has a microprocessor A7 that's running Linux. I need to read the voltage of one of the pins. The pin configuration for the Analogical to Digital conversion it's made with the Linux terminal in the STM32.

How do I write in the Python code the order to "extract" the digitalized data from the kernell?

tfcollins, 2022-10-01:
Not sure if libiio is the right place for your question. Does your ADC have an IIO driver? If you don't know you should probably reach out to ST's support.

-Travis

andresbove, 2022-10-01:
Yes, it does have. I'm going to use this configuration.

https://wiki.st.com/stm32mpu/wiki/How_to_use_the_IIO_user_space_interface#How_to_do_a_simple_ADC_conversion_using_the_sysfs_interface

tfcollins, 2022-10-02:
I would start here then: https://analogdevicesinc.github.io/libiio/v0.24/python/index.html

-Travis

andresbove, 2022-10-02:
Thank you very much!

andresbove, 2022-10-04:
If I want to use the Python Bindings in libiio for accessing and using the data of the ADC, do I must do it with the "buffer"? Following the herachy, should I create an "context" and  a "device"? Like in this code:

https://analogdevicesinc.github.io/libiio/v0.20/python/_modules/iio_readdev.html#DataReader

Now, how do I tell the application what to read? Is the "device" the STM32 ADC pin?

tfcollins, 2022-10-04:
IIO apps follow the hierarchy by creating a context, then find a device, and manipulate those device's channels.

Here is a basic example of an Analog Devices transceiver https://github.com/analogdevicesinc/plutosdr_scripts/blob/master/pluto.py

You should follow a similar flow.

-Travis

andresbove, 2022-10-04:
So, for the context creation can I use `ctx=iio.LocalContext` and link all the devices that I'm going to use to that context? Because I think that I don't need a context working with an IP address

And for the "device part" could be something like this? 
`ctrl = ctx.find_device("48003000.adc:adc@0")`
Following the below picture:

[image]

tfcollins, 2022-10-04:
```ctx = iio.Context()``` is the same as ```ctx = iio.LocalContext()```.

find_device looks at the name device attribute. Which would be ```48003000.adc:adc@0``` is your case.

dNechita, 2024-12-03:
Closing this as no further questions were asked for quite some time.

closed 2024-12-03 by dNechita