analogdevicesinc/libiio #911: make iiod dynamic on kernel (sysfs) changes

rgetz, 2022-10-05:
Today, when the kernel changes sysfs (due to a hardware, or device tree overlay change), iiod needs to be restarted manually. This sort of stinks, and should be fixed for libiio 1.0

It would be better if:
- iiod was using poll() on sysfs (greg says not to use inotify - https://www.spinics.net/lists/linux-fsdevel/msg73924.html but that might be outdated now). Maybe @mhennerich might have some thoughts.
- when sysfs changes - figure out what to do. If a device removes/adds attributes, it should likely return errors for all future calls, until the context is closed/re-opened. If the device doesn't change - it's likely OK to continue to work on it. I don't know if you chance device "A" inside a context - if all devices should be closed, or not.

This is based on a discussion that @pcercuei and I were having at GNU Radio Conference.

-Robin

mhennerich, 2022-10-05:
Well that's what udev is for. 
Please see uevent: https://documentation.suse.com/sles/12-SP4/html/SLES-all/cha-udev.html

You can write a simple udev rule which will restart IIOD automatically, when a device get's removed or added.
The only thing is that we need to inform the remote side when this has happened.
Exchange a new context and handle error conditions.

rgetz, 2022-10-06:
I think making udev killing iiod at random times when devices can be attached to them - and then just letting things time out / would be problematic from a user standpoint. I get that is what we have today - but we should hopefully be getting better...

iiod needs to notice things, and then turn off the existing connections off gracefully (or as graceful as they can be). There is a difference between getting an error saying "device changed, re-open context" and all the devices just randomly disappearing...

-Robin

nunojsa, 2022-10-06:
> I think making udev killing iiod at random times when devices can be attached to them 

Not sure about what this but I would suspect that you can control how udev can notify iiod on device unplug/plug (I guess we just care about any change on sysfs) without having to kill it (using udev could save us from polling on sysfs). I'm thinking on a specific signal for instance or something with pipes/sockets...

Anyways, my 5 cents on this one is that we should do this as simple as possible. Ultimately, it should be applications to handle this (reloading contexts). IIOD/libiio should just have a mechanism to let them know that things have changed. Handling this things dynamically is extremely hard (not even the kernel does it right...), so things can become extremely complicated very quickly...

rgetz, 2022-10-17:
> If IIOD gets killed then the applications would most likely get -EPIPE on all operations,

but - I don't think this happens if timeouts are set to long or infinite? does it? I have definitely seen an iio client hang indefinitely when iiod dies at the wrong time.

-Robin

pcercuei, 2022-10-17:
It should happen even when timeouts are set to infinite.

nunojsa, 2022-10-19:
> It should happen even when timeouts are set to infinite.

I wonder what happens if we receive SIGPIPE during `poll()`...

```
do {
        ret = poll(pfd, 2, timeout);
} while (ret == -1 && errno == EINTR);
```

I guess we would just continue to poll?  From the top of my head I don't think poll() will return right away on a socket with a closed connection... But I did not tried it though.

If the above is true, on linux we could (maybe) get way with registering SIGPIPE with `signalfd()`... No idea on other platforms

pcercuei, 2022-10-19:
@nunojsa you shouldn't actually get a SIGPIPE, since we poll() every time before reading or writing data to the sockets.

nunojsa, 2022-10-19:
But if the iiod get's killed while the client is on a poll() call, won't the client get SIGPIPE? Or is it something else? Long time that I don't play with poll and friends so I honestly don't remember what happens when it's the server that ends the connection... Will we just receive 0 from `recv()` as well?

pcercuei, 2022-10-19:
No, the poll() would get a `POLLHUP` flag set in revents.

`recv()` would get a SIGPIPE, yes, unless the MSG_NOSIGNAL flag is used, which is what IIOD does to avoid getting signals itself, btw.

pcercuei, 2022-10-19:
Actually recv() does not trigger a SIGPIPE, when the connection has been closed by the remote or the network link is lost, it will read whatever is left in the kernel buffer, and eventually return 0, which signals the "end of file".

`send()` however can actually give us a SIGPIPE, since wait_cancellable() does not return -EPIPE when POLLHUP is set. I'll send a patch.

nunojsa, 2022-10-20:
> send() however can actually give us a SIGPIPE, since wait_cancellable() does not return -EPIPE when POLLHUP is set. I'll send a patch.

Yes, but I wonder if we still receive the signal when wait_cancellable() is called. I mean, of course we'll receive it but likely we will receive -EINTR in the poll() call and then return POLLHUP. I think we will still get -EPIPE in the next send() call, so it should be fine.

I also wonder if we shouldn't treat POLLHUP and POLLERR differently in wait_cancellable()?

rgetz, 2023-08-03:
I think this can be closed since #975 was merged - and it should be straightforward to do this with udev now...