analogdevicesinc/libiio #916: iio_info generates "Unable to create Local IIO context : No such file or directory (2)" message

warnes, 2022-10-28:
On Ubuntu 22.04, with libiio installed from the current github master branch:
```
$ sudo iio_info -s
Library version: 0.24 (git tag: e598e4b)
Compiled with backends: local xml ip usb
Unable to create Local IIO context : No such file or directory (2)
Available contexts:
	0: 192.168.2.1 (Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)), serial=1044734c9605000211000700b41b503492 [ip:pluto-green.local]
	1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=1044734c9605000211000700b41b503492 [usb:3.39.5]
```

tfcollins, 2022-10-28:
That warning is expected if you have no local iio devices. It can be ignored.

warnes, 2022-10-28:
@tfcollins That's what's odd.  I *do* have a local iio device connected, as you can see in the output I included.  Here it is with *two* attached PlutoSDR's:
```
$ iio_info -s
Library version: 0.24 (git tag: e598e4b)
Compiled with backends: local xml ip usb
Unable to create Local IIO context : No such file or directory (2)
Available contexts:
	0: 192.168.2.1 (Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)), serial=1044734c9605000211000700b41b503492 [ip:pluto-green.local]
	1: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=1044734c9605000211000700b41b503492 [usb:3.59.5]
	2: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=1044734c9605000315000600a42c505f26 [usb:3.60.5]
````

tfcollins, 2022-10-28:
A local context means the driver is running in your host's kernel. In your case the drivers are running inside the ARM of the Plutos

warnes, 2022-10-28:
Ahh.  Thanks for the clarification.

closed 2022-10-28 by warnes