analogdevicesinc/libiio #923: Conflicts with `libiio-dev` and ` libad9361` Debian packages

Jachimo, 2022-12-27:
This is hopefully just a trivial packaging problem, but I have run into an issue where, after installing `Ubuntu-22.04_latest_master_libiio.deb` (which provides the package `libiio` at version `0.24.ga08ad5f`), I am blocked from installing `libiio-dev` through normal channels.  I am fairly certain an identical problem exists with ` libad9361-0` as well, had it not been installed already.

This is problematic because `libiio-dev` is a dependency of several projects, [SDR++ among them](https://github.com/AlexandreRouma/SDRPlusPlus). So a lot of users are going to want to install that.

Is there an easy workaround to update `libiio` from Github to the latest but still have `libiio-dev` installed?  Or a way to install a matching version of `libiio-dev` from Github that will match the `libiio` .deb package version?

For reference, here is what my system shows for `sudo apt search libiio`:

```
$ sudo apt search libiio
Sorting... Done
Full Text Search... Done
iiod/jammy 0.23-2 amd64
  Daemon to share IIO devices on the network

libad9361-0/jammy,now 0.2-1 amd64 [installed,automatic]
  Library of functions specific to the Analog Devices AD9361

libad9361-dev/jammy,now 0.2-1 amd64 [installed]
  Development files specific to the Analog Devices AD9361

libgnuradio-iio3.10.1/jammy 3.10.1.1-2 amd64
  IIO blocks for GnuRadio

libiio/now 0.24.ga08ad5f amd64 [installed,local]
  libiio built using CMake

libiio-dev/jammy 0.23-2 amd64
  libiio development files

libiio-utils/jammy 0.23-2 amd64
  Miscellaneous libiio utilities

libiio0/jammy 0.23-2 amd64
  Library for interfacing with IIO devices

libiio0-doc/jammy,jammy 0.23-2 all
  libiio documentation

python3-libiio/jammy,jammy 0.23-2 all
  Python bindings for libiio
```

And if I try to install `libiio-dev` from the Jammy repo, for instance because I want to run SDR++, I get this (entirely reasonable) error message:

```
$ sudo apt install libiio-dev
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
You might want to run 'apt --fix-broken install' to correct these.
The following packages have unmet dependencies:
 libad9361-0 : Depends: libiio0 but it is not going to be installed
 libiio-dev : Depends: libiio0 (= 0.23-2) but it is not going to be installed
E: Unmet dependencies. Try 'apt --fix-broken install' with no packages (or specify a solution).
```

Does `libiio-dev` come from this Git repository? Or is there someone else who packages it?

Jachimo, 2022-12-27:
Just for thoroughness, this is what you get if you try to install `libiio0` using APT, after having installed the .deb package from Github:

```
$ sudo apt install libiio0
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following NEW packages will be installed:
  libiio0
0 upgraded, 1 newly installed, 0 to remove and 36 not upgraded.
83 not fully installed or removed.
Need to get 0 B/52.4 kB of archives.
After this operation, 145 kB of additional disk space will be used.
(Reading database ... 213661 files and directories currently installed.)
Preparing to unpack .../libiio0_0.23-2_amd64.deb ...
Unpacking libiio0:amd64 (0.23-2) ...
dpkg: error processing archive /var/cache/apt/archives/libiio0_0.23-2_amd64.deb (--unpack):
 trying to overwrite '/usr/lib/x86_64-linux-gnu/libiio.so.0', which is also in package libiio 0.24.ga08ad5f
Errors were encountered while processing:
 /var/cache/apt/archives/libiio0_0.23-2_amd64.deb
E: Sub-process /usr/bin/dpkg returned an error code (1)
```

Again I'm not saying that this is _wrong_ (it's exactly the right behavior—the newer version shouldn't be overwritten so it's good the process stops), but it's a bit of an odd Catch-22 situation because it's not clear where to get `libiio-dev` once you have installed updated `libiio`.

pcercuei, 2023-01-09:
Hi @Jachimo, sorry for replying so late - holidays etc.

Maybe SoapySDR should search for `libiio || libiio-dev`, then it should work.

rgetz, 2023-02-03:
I think @maitbot might provide some insight as well.

-Robin

maitbot, 2023-02-04:
The github built .debs do not follow policy.
I split libiio from libiio-dev (and include the soversion in the library package name libiio0) because following the policy allows for smooth upgrades without recompiling all dependent packages at once.

Also, this issue has some confusion over build-time dependencies (*-dev) vs run-time dependencies.

Read more at https://www.debian.org/doc/debian-policy/ch-sharedlibs.html

commit 49dc23d3 (2023-06-20): cmake: Handle conflict between generated .deb and Debian package

Update the generated .deb's metadata to enforce that it cannot be
installed alongside the version of libiio shipped in the Debian
repositories.

Fixes #923.
PR #988 (2023-06-21): cmake: Handle conflict between generated .deb and Debian package
Update the generated .deb's metadata to enforce that it cannot be installed alongside the version of libiio shipped in the Debian repositories.

Fixes #923.
commit 8ab3029c (2023-06-21): cmake: Handle conflict between generated .deb and Debian package

Update the generated .deb's metadata to enforce that it cannot be
installed alongside the version of libiio shipped in the Debian
repositories.

Fixes #923.
closed 2023-06-21 by pcercuei