analogdevicesinc/libiio #926: context description with serial backend

buha, 2023-01-20:
1. Given the following string in xml:

```
<context name="xml" version-major="0" version-minor="24" version-git="c4498c2" description="no-OS master-9271c4ee2-modified" >
```
When using the serial backend, I get the following context name and description from libiio:
```
>>> serctx.name
'serial'
>>> serctx.description
'/dev/ttyACM0: DAPLink CMSIS-DAP - 0425170267cb0dfe00000000000000000000000097969906'
```

2. Given the following string in xml:

```
<context name="xml" version-major="0" version-minor="24" version-git="c4498c2" description="Linux (none) 5.4.0-gfd4058d9a174 #121 SMP PREEMPT Wed Apr 14 08:46:13 CEST 2021 armv7l" >
```
When using the usb backend, I get the folloing context name and description from libiio:
```
>>> usbctx.name
'usb'
>>> usbctx.description
'Linux (none) 5.4.0-gfd4058d9a174 #121 SMP PREEMPT Wed Apr 14 08:46:13 CEST 2021 armv7l'
```

At point 2, the description is as expected.
At point 1 however, the description contains serial port info instead of the xml description attribute.

I've traced this to this function:
https://github.com/analogdevicesinc/libiio/blob/e9c801112c28e27ada054bae5ed260d4618f5f21/serial.c#L113

I think it would be better to have consistent behavior across backends and populate context description in libiio with the content from xml and not from libserialport. (Perhaps populate with info from libserialport if the description attribute is missing from xml, but this is optional).

pcercuei, 2023-01-20:
This should already be fixed for the next Libiio version in the dev branch.

buha, 2023-01-20:
oh really ? I will give it a try

pcercuei, 2023-01-20:
It's not an error, it's a warning :)

buha, 2023-01-23:
Oh yeah, my bad...

So it's ok now, I can build/install the dev branch. However I get  Invalid argument (22) when trying to use the new libiio:

```
$ iio_genxml -u serial:/dev/ttyACM0,57600
ERROR: Unable to create IIO context serial:/dev/ttyACM0,57600: Invalid argument (22)
```

Is there anything that changed in xml parsing ? Does the `dev` branch expect something in the xml which was not there previously ?

Here's my xml for reference:
```
<?xml version="1.0" encoding="utf-8"?>
<!DOCTYPE context [
<!ELEMENT context (device | context-attribute)*>
<!ELEMENT context-attribute EMPTY>
<!ELEMENT device (channel | attribute | debug-attribute | buffer-attribute)*>
<!ELEMENT channel (scan-element?, attribute*)>
<!ELEMENT attribute EMPTY>
<!ELEMENT scan-element EMPTY>
<!ELEMENT debug-attribute EMPTY>
<!ELEMENT buffer-attribute EMPTY>
<!ATTLIST context name CDATA #REQUIRED description CDATA #IMPLIED>
<!ATTLIST context-attribute name CDATA #REQUIRED value CDATA #REQUIRED>
<!ATTLIST device id CDATA #REQUIRED name CDATA #IMPLIED>
<!ATTLIST channel id CDATA #REQUIRED type (input|output) #REQUIRED name CDATA #IMPLIED>
<!ATTLIST scan-element index CDATA #REQUIRED format CDATA #REQUIRED scale CDATA #IMPLIED>
<!ATTLIST attribute name CDATA #REQUIRED filename CDATA #IMPLIED>
<!ATTLIST debug-attribute name CDATA #REQUIRED>
<!ATTLIST buffer-attribute name CDATA #REQUIRED>]>
<context name="xml" description="no-OS microwave-b0fd22e4a-modified" >
	<device id="iio:device0" name="hmc6300">
		<channel id="temp" type="input" >
			<attribute name="raw" filename="in_temp_raw" />
		</channel>
		<attribute name="enabled" />
		<attribute name="vco" />
		<attribute name="vco_available" />
		<attribute name="if_attn" />
		<attribute name="rf_attn" />
		<debug-attribute name="direct_reg_access" />
	</device>
	<device id="iio:device1" name="hmc6301">
		<channel id="temp" type="input" >
			<attribute name="raw" filename="in_temp_raw" />
		</channel>
		<attribute name="enabled" />
		<attribute name="vco" />
		<attribute name="vco_available" />
		<attribute name="if_attn" />
		<attribute name="bb_attn1" />
		<attribute name="bb_attn2" />
		<attribute name="bb_attni_fine" />
		<attribute name="bb_attnq_fine" />
		<debug-attribute name="direct_reg_access" />
	</device>
	<device id="iio:device2" name="mwc">
		<channel id="voltage" name="tx_det" type="input" >
			<attribute name="raw" filename="in_voltage_raw" />
		</channel>
		<channel id="voltage" name="rx_det" type="input" >
			<attribute name="raw" filename="in_voltage_raw" />
		</channel>
		<attribute name="tx_autotuning" />
		<attribute name="rx_autotuning" />
	</device>
</context>
```

pcercuei, 2023-01-23:
The XML looks fine - if you save it to a file test.xml, then run `iio_genxml -u xml:test.xml` it will accept it without problems.

Serial backend works *mostly* fine here, I do not get this "invalid argument" error.
What do you run on the other side? Is it regular IIOD with the serial code enabled, or tinyIIOD?

buha, 2023-01-25:
I am using TinyIIOD in no-OS. The one implemented here which is a modified version of the original TinyIIOD:
https://github.com/analogdevicesinc/no-OS/blob/master/iio/iiod.c

I tried what you suggested on `dev` branch, iio_genxml works when provided with test.xml input file but fails when getting the same thing from serial so I'm not quite sure what's going on.

buha, 2023-01-30:
So we've found out the issue was that no-OS replies with -EINVAL (-22) to the TIMEOUT IIO command.

buha, 2023-01-30:
Closing this as it's already implemented by @pcercuei on `dev` branch.

commit 89749a90 (2023-07-26): serial: Pass port name/description as context attributes

Instead of overriding the context description that comes from the XML,
pass the serial port name and description as context attributes, and
keep the context description as-is.

Fixes #926.
commit 221bfb87 (2023-07-27): serial: Pass port name/description as context attributes

Instead of overriding the context description that comes from the XML,
pass the serial port name and description as context attributes, and
keep the context description as-is.

Fixes #926.
PR #1017 (2023-07-27): serial: Pass port name/description as context attributes
Instead of overriding the context description that comes from the XML, pass the serial port name and description as context attributes, and keep the context description as-is.

Fixes #926.
commit b31f4c95 (2023-07-27): serial: Pass port name/description as context attributes

Instead of overriding the context description that comes from the XML,
pass the serial port name and description as context attributes, and
keep the context description as-is.

Fixes #926.
closed 2023-07-27 by pcercuei