analogdevicesinc/libiio #929: clear serial buffer before the context creation (PRINT)

buha, 2023-01-25:
The context for this is that there can be some junk on the serial buffer before the .xml, something like:
```
junk#$%^&<?xml version="1.0" encoding="utf-8"?>...
```

The reasons for this junk may be:
- some UARTs may toggle the Tx during a system reset, power cycle, etc.
- on no-OS we may print some human-readable text prior to launching an IIOD server in firmware

The symptoms when this happens is that the client (iio-oscilloscope for example) may need multiple Refresh clicks until the context is properly displayed. Sometimes this doesn't work and one needs to open up a serial terminal (picocom, minicom etc.) to clear the buffer and then launch the iio client.

I think libiio needs to disregard anything in the serial buffer prior to the PRINT command reply.
I may not be correct here (you know better libiio) but it seems to me that the correct place for clearing the buffer is somewhere before this line:
https://github.com/analogdevicesinc/libiio/blob/e9c801112c28e27ada054bae5ed260d4618f5f21/serial.c#L475

Let me know what do you think 

Tagging @dbogdan and @rbolboac to follow this.

pcercuei, 2023-01-25:
Libiio already clears the serial buffers before the PRINT command is sent:
https://github.com/analogdevicesinc/libiio/blob/2945479041f6f8df828f81f91de88d4271e0d588/serial.c#L455-L456

buha, 2023-01-25:
Yep, it's there.
But I did some testing and it seems it doesn't work unfortunately... 

I modified libiio like this:
```
diff --git a/iiod-client.c b/iiod-client.c
index f00de33b..2172059b 100644
--- a/iiod-client.c
+++ b/iiod-client.c
@@ -536,8 +536,23 @@ iiod_client_create_context_private(struct iiod_client *client,
 	size_t xml_len;
 	char *xml;
 	int ret;
+	char c;
+	ssize_t s;
+
+	IIO_INFO("[");
+	do {
+		s = client->ops->read(client->pdata, desc, &c, 1);
+		if (s == 1)
+			IIO_INFO("%c", c);
+		else
+			break;
+	} while(1);
+	IIO_INFO("]");
+
+	IIO_INFO("read() returned %d\n", s);
 
 	iio_mutex_lock(client->lock);
+
 	ret = iiod_client_exec_command(client, desc, cmd);
 	if (ret < 0) {
 		if (ret == -EINVAL && zstd) {
```
So essentially the code just prints leftovers characters from sp_flush() between `[...leftover chars...]` when the context is created.

These junk characters I produce from no-OS before I initialize the IIO server.
Here's what gets printed:

```
$ osc
ERROR: Unable to create Avahi DNS-SD client :Daemon not running
[��

�^�&R�Z
ERROR: sp_blocking_read_next has timedout
�E�r�]read() returned -110
ERROR: sp_blocking_read_next has timedout
[]read() returned -110
```

I would expect to get no junk characters if sp_flush() would work.

pcercuei, 2023-01-27:
I don't actually error-check `sp_flush`, maybe it fails?

Are you using hardware flow control (e.g. CTS/RTS)?

buha, 2023-01-27:
I just put a message to view its return value and it's zero:
```
sp_flush returned: 0
[��

�^�&R�Z
ERROR: sp_blocking_read_next has timedout
�E�r�]read() returned -110
ERROR: sp_blocking_read_next has timedout
[]read() returned -110
```

```
diff --git a/serial.c b/serial.c
index adf2b3c4..c286d2db 100644
--- a/serial.c
+++ b/serial.c
@@ -453,8 +453,9 @@ static struct iio_context * serial_create_context(const char *port_name,
         }
 
         /* Empty the buffers */
-        sp_flush(port, SP_BUF_BOTH);
-
+        enum sp_return ret2 = sp_flush(port, SP_BUF_BOTH);
+        IIO_INFO("sp_flush returned: %d\n", ret2);
+        
         description = serial_get_description(port);
         if (!description)
                 goto err_close_port;
```

I don't use flow control.

Do you have any means to reproduce this ?

pcercuei, 2023-01-27:
I pushed a branch `pcercuei/dev-drain-serial-input` (note: based on the *dev* branch), could you give it a try?

buha, 2023-01-30:
`dev` branch doesn't work for me for some reason, I always get Invalid argument (22) from libiio.

I've put your pcercuei/dev-drain-serial-input top commit on top of master and tried it that way but it doesn't fix the issue.

buha, 2023-01-30:
Fixed by #931

buha, 2023-07-26:
@pcercuei I've just tested 0.25-rc1 for this feature and I still don't see it in there, I haven't checked the git log of libiio, do you plan on including this in 0.25?

I have this in the xml:
```
<context name="xml" version-major="0" version-minor="25" version-git="v0.25" description="no-OS tags/wethlink-0.1.7-d3207fb6d" >
```

And iio_info shows this at Backend description string:
```
iio_info -u serial:/dev/ttyACM0,115200,8n2n
iio_info version: 0.25 (git tag:v0.25)
Libiio version: 0.25 (git tag: v0.25) backends: local xml ip usb serial
IIO context created with serial backend.
Backend version: 1.1 (git tag: 0000000)
Backend description string: /dev/ttyACM0: DAPLink CMSIS-DAP - 0409170256d9088800000000000000000000000097969906
```

buha, 2023-07-26:
closing again

closed 2023-07-26 by buha