analogdevicesinc/libiio #940: ADUCM3029 runs IIO and reports an error:Compiled with backends: xml ip usb serial Unable to create Local IIO context : Function not implemented No IIO context found.

smarthobee612, 2023-02-22:
[光学手势传感器演示 （带 EVAL-CN0569-PMDZ） [ADI公司维基] (analog.com)](https://wiki.analog.com/resources/eval/user-guides/circuits-from-the-lab/cn0569/sw)

PS D:\pyadi-iio\examples\cn0569> iio_info -s
Library version: 0.21 (git tag: 565bf68)
Compiled with backends: xml ip usb serial
Unable to create Local IIO context : Function not implemented
No IIO context found.

PS D:\pyadi-iio\examples\cn0569> iio_info.exe -s
Library version: 0.24 (git tag: c4498c2)
Compiled with backends: xml ip usb serial
Unable to create Local IIO context : Function not implemented (40)
No IIO context found.

PS D:\pyadi-iio\examples\cn0569> iio_attr -a -C
No IIO context found.

pcercuei, 2023-02-22:
I believe you are connected through UART?

Libiio's serial backend does not support scanning, so it won't auto-detect the board.

However you should be able to use iio_info with the board, if you know the COM port.
For instance: `iio_info -u serial:COM1,57600n8`

smarthobee612, 2023-02-23:
Thanks! pcercuei! The ADCUM3029 development board communicates with the MCU through the onboard DAP-link serial port. At present, according to your method, the test is as follows. My computer recognizes that the serial port of the development board is COM4, and still reports an error. Does the iio service run on the MCU? Only when IIO is running can the IIO device be discovered on the PC side?
PS C:\Windows\system32> iio_info -u serial:COM4,115200n8
Library version: 0.23 (git tag: c14a0f8)
Compiled with backends: xml ip usb serial
ERROR: Invalid number of bits
ERROR: Bad URI: 'serial:COM4,115200n8'
Unable to create IIO context serial:COM4,115200n8: Invalid argument (22)
I tried different speeds but got an error：
PS C:\Windows\system32> iio_info -u serial:COM4,57600n8
Library version: 0.23 (git tag: c14a0f8)
Compiled with backends: xml ip usb serial
ERROR: Invalid number of bits
ERROR: Bad URI: 'serial:COM4,57600n8'
Unable to create IIO context serial:COM4,57600n8: Invalid argument (22)
PS C:\Windows\system32> iio_info -u serial:COM4,9600n8
Library version: 0.23 (git tag: c14a0f8)
Compiled with backends: xml ip usb serial
ERROR: Invalid number of bits
ERROR: Bad URI: 'serial:COM4,9600n8'
Unable to create IIO context serial:COM4,9600n8: Invalid argument (22)

smarthobee612, 2023-02-23:
I read the iio data of ADI, tried the command, and still reported an error, ERROR: sp_blocking_read_next has timedout
ERROR: READ LINE: -138
Where exactly is this problem? I haven't figured it out yet.
PS C:\Windows\system32> iio_info -u serial:COM4,115200
Library version: 0.23 (git tag: c14a0f8)
Compiled with backends: xml ip usb serial
ERROR: sp_blocking_read_next has timedout
ERROR: READ LINE: -138
Unable to create IIO context serial:COM4,115200: timed out (138)

rgetz, 2023-02-23:
I will update the man page, for now, check the c-code API.

http://analogdevicesinc.github.io/libiio/v0.24/libiio/group__Context.html#gafdcee40508700fa395370b6c636e16fe

here is an expanded version:

```
Serial backend, "serial:"
Requires (in this order, but you can stop at any time to use the defaults):
  * a port (/dev/ttyUSB0, no default - this is required)
  * a comma
  * baud_rate (110, 300, 600, 1200, 2400, 4800, 9600, 14400, 19200, 38400, 57600, 115200 [default], 128000 or 256000)
  * a comma
  * serial port configuration (default is 8 data bits, no parity, 1 stop bits, no flow control).
     * data bits (5, 6, 7, 8 [default], or 9)
     * parity ('n' none [default], 'o' odd, 'e' even, 'm' mark, 's' space)
     * stop bits (1 [default], or 2)
     * flow control ('\0' none [default], 'x' Xon Xoff, 'r' RTSCTS, or 'd' DTRDSR)
For example "serial:/dev/ttyUSB0,115200" or "serial:/dev/ttyUSB0,115200,8n1"
```

rgetz, 2023-02-23:
are you sure `COM4` is the serial port that the device is connected to? check device manager to find your serial port.

smarthobee612, 2023-02-23:
Yes, it is COM4 interface seen in the device manager.
The program running in ADCUM3029 is DEMO: https://wiki.analog.com/resources/eval/user-guides/circuits-from-the-lab/cn0569/sw

rgetz, 2023-02-24:
I don't think this is a problem with libiio - your best bet is to go to ADI's support forms for the CN0569.
I think (but someone can correct me) it's https://ez.analog.com/reference-designs/

smarthobee612, 2023-02-27:
thanks，rgetz！There is another question. Is the iio service running in the MCU or embedded system?

pcercuei, 2023-03-01:
The ADICUP board should be running a firmware that implements https://github.com/analogdevicesinc/libtinyiiod/.
If you used https://github.com/analogdevicesinc/no-OS/tree/master/projects/iio_adpd1080 as the firmware, it does implement it; so unless there is a bug with the software - it should be available to communicate over UART.

If you want to test if the firmware is working properly, you could try to use a UART console on your PC, and connect to the correct COM port. The commands are sent as regular text, so you can manually write e.g. `PRINT` in the console, and the board should answer with a XML string.

If you get garbage data, your UART settings (baudrate, stop-bits etc.) are probably wrong. If you don't receive anything from the board, either the COM port is not the right one, or the firmware on the board does not run properly.

The iio service is started here: https://github.com/analogdevicesinc/no-OS/blob/master/projects/iio_adpd1080/src/main.c#L430
So if it turns out that the iio service is not started, it probably means that the main() function exited early.

closed 2023-08-03 by pcercuei