analogdevicesinc/libiio #941: Failed to create iio buffer

panxishan, 2023-02-24:
Via iio_ device_ create_ Buffer () creates a buffer to read the information of the gyroscope's iio device. There is a segment error, error code 22. What is the reason

pcercuei, 2023-02-24:
Error code 22 is "invalid value". What hardware and Linux driver are you using, and how do you call iio_device_create_buffer?

panxishan, 2023-02-25:
> Error code 22 is "invalid value". What hardware and Linux driver are you using, and how do you call iio_device_create_buffer?

I wrote an iio driver test for mpu6050 and created a buffer in qt through the (struct iio_buffer * buf=iio_device_create_buffer (dev, 1024, false) function

pcercuei, 2023-03-01:
I would think that your mpu6050 driver does not support the buffer interface. It does not really need to, if the sample rate is very low (which is typically the case on a gyroscope).

Instead, you should be able to use `channel.find_attribute("raw").read()` to get a snapshot of the channel's value at a given time.

dNechita, 2024-12-03:
Closing this as the provided answer is reasonable, considering the details of the issue that were shared.

closed 2024-12-03 by dNechita