analogdevicesinc/libiio #948: ad7192 driver: Unable to create local context: Invalid argument (22)

roundby, 2023-03-27:
Context creation fails with this very same issue of #73, but the already applied patch does not solve the issue.
Both `iio_create_local_context()` and iiod fail with the same error.

[ad7192 driver](https://github.com/torvalds/linux/blob/master/drivers/iio/adc/ad7192.c) defines extended attribute "shorted" for channel 2, a configuration used to measure noise on input:

`AD719x_SHORTED_CHANNEL(3, 2, AD7192_CH_AIN2P_AIN2M)`

Under /sys/bus/iio/devices/iio:device0/scan_elements the driver adds files named in_voltage2_shorted_en, in_voltage2_shorted_index and in_voltage2_shorted_type.

`handle_protected_scan_element_attr()` function is called with attr "shorted_type", the final else catches unsupported attribute and returns EINVAL (the context cannot be created).

At the lowest level, `get_channel_id()` computes channel id from file names between second and third underscore characters (in_voltage2_shorted_index gives voltage2)
`get_short_attr_name()` looks for the string after the second underscore character (in_voltage2_shorted_index gives shorted_index)

We are able to properly run iiod by altering the behaviour of these two functions

`get_channel_id() `returns string between first and last underscore
`get_short_attr_name()` returns string after last underscore

While this might always be acceptable for protected attributes, where any attribute name different from "type", "en" or "index" would cause the context creation to fail, I doubt it could be universally applicable.

I kindly ask you a feedback about the pattern adopted for channel naming and suggestions for a viable solution.

In the meantime here are the applied changes for reference:

```patch
--- a/local.c     2023-03-24 13:16:51.590388000 +0100
+++ b/local.c     2023-03-24 13:18:11.346997000 +0100
@@ -1161,7 +1161,7 @@ static char * get_channel_id(const char
      size_t len;
 
      attr = strchr(attr, '_') + 1;
-     ptr = strchr(attr, '_');
+     ptr = strrchr(attr, '_');
      if (find_channel_modifier(ptr + 1, &len) != IIO_NO_MOD)
            ptr += len + 1;
 
@@ -1176,10 +1176,9 @@ static char * get_channel_id(const char
 
 static char * get_short_attr_name(struct iio_channel *chn, const char *attr)
 {
-     char *ptr = strchr(attr, '_') + 1;
      size_t len;
 
-     ptr = strchr(ptr, '_') + 1;
+     char *ptr = strrchr(attr, '_') + 1;
      if (find_channel_modifier(ptr, &len) != IIO_NO_MOD)
            ptr += len + 1;
```

pcercuei, 2023-03-29:
Hi @roundby,

Yes I am aware of the problem. This is likely to be a kernel issue - we are looking it up.

Long story short, the kernel driver creates two different channels using the "in_voltage2" identifier for the AIN2 input; one shorted and the other not. Then in sysfs, you get the following files:
```
in_voltage2_en
in_voltage2_index
in_voltage2_shorted_en
in_voltage2_shorted_index
```

Libiio detects this as a single `in_voltage2` channel with attributes "en", "index", "shorted_en", "shorted_index", because it assumes that identifiers (the "in_voltageX" part) are associated to one and only one channel.

I'll keep you posted about how we resolve this situation.

roundby, 2023-03-29:
Thank you @pcercuei for your answer.

In the hope this could be helpful, I add here a few notes.

- AD7192 provides a specific channel configuration where both positive and negative inputs are connected to AIN2. Hence the resulting file names are not inherently wrong.
- Channel voltage2_shorted ID is created by Linux IIO subsystem upon channel registration. Extended attribute ("shorted") is added to structure `iio_chan_spec` by purpose on that specific channel configuration; the field is empty for any other channel.

I am not confident with the IIO channel naming conventions and limitations, but it looks like extended attributes are not supported by libiio, at least on channel definitions.

> Libiio detects this as a single `in_voltage2` channel with attributes "en", "index", "shorted_en", "shorted_index", because it assumes that identifiers (the "in_voltageX" part) are associated to one and only one channel.

I do not see any assumption about the the channel ID when computing the attributes of the protected elements. It's just a matter of computing the string tokens between underscore characters. The failure happens before the channel ID uniqueness is verified. It would also happen if the channel were given ID voltage2-voltage2_shorted.
The invalid extended attribute verification happens at a very early local context creation stage.

BTW, the library is behaving as expected after we forced channel ID to be computed as voltage2_shorted for that specific channel.

pcercuei, 2023-03-30:
This patch was sent to the upstream kernel developers to address the issue: https://github.com/analogdevicesinc/linux/commit/ab01420b27caf70f81a9ba9a58f5db633f03c63b

rgetz, 2023-03-31:
@pcercuei : Is there a validator that should be in the kernel code somewhere? or instructions to use libiio/local?

pcercuei, 2023-06-21:
The kernel patch will be in v6.4, and will be backported to older kernel versions as well. So I consider this issue fixed.

closed 2023-06-21 by pcercuei