analogdevicesinc/libiio #967: Unable to create local context when /sys/firmware isn't available

jadonk, 2023-05-08:
I'm running libiio in a docker container that has full access to /dev/iio* and /sys/bus/iio/*, but /sys/firmware is not mapped.

```
# iio_info 
Library version: 0.21 (git tag: v0.21)
Compiled with backends: local xml ip usb serial
ERROR: Unable to stat file: No such file or directory
Unable to create Local IIO context : No such file or directory
```

The error message isn't particularly useful, but I was able to find that the issue is a symlink into /sys/firmware...

```
# ls -l /sys/bus/iio/devices/iio\:device1/of_node 
lrwxrwxrwx 1 root 999 0 Aug  7  2022 /sys/bus/iio/devices/iio:device1/of_node -> ../../../../../../../../../firmware/devicetree/base/bus@f0000/bus@4000000/spi@4b10000/adc@1
```

In this case, `/sys/firmware` is an empty directory, resulting in these lines failing in the library:
https://github.com/analogdevicesinc/libiio/blob/master/local.c#L1750-L1751

I believe the proper behavior should be to simply `continue`, rather than `goto out_close_dir`. This will result in the callback running on files that can be read and ignoring files that cannot be read. I believe that is the desired behavior, not simply exiting the application needlessly.

pcercuei, 2023-05-08:
We should probably ignore `of_node` in particular, or in general ignore all symbolic links...

rgetz, 2023-05-11:
@pcercuei - we should be able to catch that with something like....

```
if (S_ISLNK(st.st_mode)) continue;
```
?

and then we would not follow any symlinks...

pcercuei, 2023-05-11:
Sure, but there might be real use cases where you have symlinks in there. Having symlinks in sysfs is not uncommon. Having broken symlinks is, so I'm wondering whether it is something that we should handle, or @jadonk's kernel is somewhat broken.

jadonk, 2023-05-11:
In what case should we fail because we cannot read an unnecessary file? If we cannot read a file, why can we not simply ignore it. 

Swapping the goto with a continue works great. I think trying to over-analyze what to do with something you cannot anticipate and do not need is a waste of mental and computing energy. 

What am I missing here?

jadonk, 2023-05-11:
My kernel isn't broken, I'm running inside a docker container that has an idea that /sys/firmware is dangerous. Docker is brain dead when it comes to its fake security, but let's not have docker decide we cannot run libiio over some completely unnecessary access. libiio doesn't need to access /sys/firmware for its basic function, so let's just enable it to work when it can.

pcercuei, 2023-05-11:
Silently ignoring errors is not a solution. I'm just wondering why the kernel has a dead link in its sysfs filesystem. If that's a particular kernel configuration that causes it, it should be reported as a kernel bug.

pcercuei, 2023-05-11:
@jadonk Does adding `"of_node"` to the `device_attrs_blacklist[]` array in local.c fixes it for you?

jadonk, 2023-05-18:
No idea. The continue works great.

rgetz, 2023-05-31:
@jadonk - I get the feeling that we are not going to add your suggested solution to the upcoming release. Can you test @pcercuei solution and see if that fixes your issue? and then we can look at adding that...

Thanks
-robin

jadonk, 2023-05-31:
Sorry, the whitelist makes no sense to me. I'd rather carry a patch I understand than encourage use of a patch that is unreasonably complicated.

pcercuei, 2023-05-31:
The patch would be literally one line. I wouldn't call that "unreasonably complicated".

Also, we absolutely do not want to hide errors. Your suggestion might work for you, but it is not something that we want to do in the official repo.

rgetz, 2023-06-03:
Ok - so I guess we can close this as won't fix, because we have no system to replicate the error on.

closed 2023-06-07 by pcercuei

alexmaurer-madis, 2025-01-08:
I'm facing the exact same problem on a Toradex iMX6ULL board with libiio 0.24
I cannot create local context, errno is 2.

The system is running Docker and /sys/firmware is empty.

output of iio_info is 

```
Library version: 0.24 (git tag: v0.24)
Compiled with backends: local xml ip usb
ERROR: Unable to stat file: No such file or directory (2)
Unable to create Local IIO context : No such file or directory (2)
```

alexmaurer-madis, 2025-04-15:
The problem can be solved by launching the container under docker with

```yaml
security_opt:
   - systempaths=unconfined

```