analogdevicesinc/libiio #974: Device removal causes empty index within |ctx->nb_devices|

HarveyCHYang, 2023-06-05:
Hello,

I'd like to know if the current context supports device removals. It seems that `iio_context.nb_devices` implies devices with id 0~(nb_devices-1) should all exist, while it might not be true if a pluggable device with id != (nb_devices-1) is removed.

Should we support this use case? Thanks!

pcercuei, 2023-06-05:
Hi @HarveyCHYang,

Libiio does not handle device removals. It would need to be completely redesigned from scratch.

However... I believe that if you remove a device, all Libiio functions that manipulate this IIO device (or channel, or buffer) will return `-ENOENT` (aka. "no such file or directory), because the Linux side cannot access the device's sysfs nodes anymore.
I did not test this, however.

HarveyCHYang, 2023-06-06:
Thanks Paul!

Do you think it makes sense to return nullptr in `iio_context_get_device` [1] if the device (file/directory) doesn't actually exist, by adding a simple check though?

[1]: https://github.com/analogdevicesinc/libiio/blob/master/context.c#L285

pcercuei, 2023-06-06:
Unfortunately that would break the ABI :(
Some applications call `iio_context_get_device` in a loop with an incrementing identifier, until it returns NULL. What you suggest would break these applications.

HarveyCHYang, 2023-06-09:
I forgot to mention: in our use cases (CrOS), we would reload to get a new iio context upon an iio device being unplugged. But I guess that doesn't help, right? As some applications might create a context with empty indices...?

HarveyCHYang, 2023-06-09:
Ah we have our own logic to support IPC clients with plugged/unplugged devices, so don't really need to disconnect all clients (I assume your talking about other contexts than the local context. In CrOS we only use local context in a service though, and other sensor clients would need to talk to the service with another IPC, called Mojo, to get iio data).

We're just wondering if there's any plan for the unplugged device support in libiio, so that we can save some hacks in our service :)

pcercuei, 2023-06-09:
Then sorry, it is not planned.

HarveyCHYang, 2023-06-13:
Thanks for confirming, Paul :)

closed 2023-06-13 by HarveyCHYang