analogdevicesinc/libiio #983: Help using libiio: querying stale data instead of most recent readings

p-b-j, 2023-06-13:
Hi all,

I'm attempting to query voltage information from an ads7955 using libiio. When requested, I want the application to print the latest reading (one sample) from each of 4 voltage channels.

With my current implementation, when I change the input voltage for a particular channel, it takes multiple new queries (~5) before I see a correct reading for the new voltage (not a timing issue - I must issue multiple queries before the value updates). When I query via sysfs, I see the new reading immediately after changing the voltage. This makes me think my libiio code is reading buffered data from the sensor instead of the latest reading.

My guess is that I'm using the libiio library incorrectly. Below is some pseudo code that hopefully should be sufficient to communicate what I am doing.

Specifically I'm wondering if my assumption that `iio_buffer_refill` completely refills the buffer with new samples is correct or not. I'm also open to any suggestions for better ways to query the samples on-demand from the channels.

```
- Initialization steps (done once during initial setup)
    - Create a local context
    - Create an hrtimer trigger w/sampling_frequency=100
    - Find ads7955 device w/context
        - assign trigger to device
    - For each voltage channel in use (currently 4 channels):
        - Find and enable the channel
    - Initialize a buffer w/space for 64 samples using iio_device_create_buffer

- Query steps, done once each time the user requests:
    - refill buffer w/new samples using iio_buffer_refill
    - For each channel (currently 4 channels):
        - Read a single 2-byte sample using iio_channel_read
        - Scale the sample appropriately and report as voltage
```

I can provide more information/real code snippets where it would be helpful. Thanks for all the work y'all do maintaining the library.

p-b-j, 2023-06-13:
Providing an update - destroying the old buffer and creating a new buffer every query seems to retrieve the correct/updated adc readings. This seems inefficient though - is there a way to completely refresh the buffer w/out destroying it and creating a new one?

tfcollins, 2023-06-15:
Try setting the kernel buffers to 1. Otherwise data will get queued at buffer creation and not on fill requests for the first set of buffers

p-b-j, 2023-06-19:
Thanks appreciate the tip. I switched to setting the device kernel buffers to 1 before creating the device buffer and it improved things, but I'm still querying stale data after calling `iio_buffer_refill` (the improvement is that only the first query has stale data instead of the first 5). Destroying/recreating the buffer is still the only thing that has retrieved updated values on the first query.

p-b-j, 2023-06-21:
Thanks Travis, having that info/background knowledge helps (as does knowing there isn't a better way to accomplish my use case). I'll use those methods, marking as closed as I have no further questions.

closed 2023-06-21 by p-b-j