analogdevicesinc/libiio #984: Libiio 1.0 does not compile with WITH_AIO=0 and WITH_IIOD_USBD=1

catkira, 2023-06-14:
it gives me this error:
```bash
[ 94%] Built target iio_rwdev
[ 96%] Linking C executable iio_stresstest
[ 98%] Linking C executable iiod
[100%] Linking C executable iio_attr
/usr/bin/ld: CMakeFiles/iiod.dir/iiod.c.o: in function `main':
/home/benja/git/libiio/iiod/iiod.c:531: undefined reference to `start_usb_daemon'
[100%] Built target iio_stresstest
collect2: error: ld returned 1 exit status
make[2]: *** [iiod/CMakeFiles/iiod.dir/build.make:227: iiod/iiod] Error 1
make[1]: *** [CMakeFiles/Makefile2:466: iiod/CMakeFiles/iiod.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Built target iio_attr
make: *** [Makefile:136: all] Error 2
```
is this configuration supposed to work? if yes, `usbd.c` needs to be added to compilation.

rgetz, 2023-06-19:
I think this is on purpose.

WITH_IIOD_USBD is inside the WITH_AIO option

https://github.com/analogdevicesinc/libiio/blob/master/iiod/CMakeLists.txt#L34-L43

catkira, 2023-06-19:
in that case it would be good to give an error messsge about the illegal parameter combination

rgetz, 2023-06-19:
yeah, I had mentioned that to Paul awhile ago, I can make a PR for it.

commit 44804226 (2023-12-07): cmake: ensure WITH_AIO=0 and WITH_IIOD_USBD=1 fails

at cmake time, not at link time.

Turns this:
/usr/bin/ld: CMakeFiles/iiod.dir/iiod.c.o: in function `main':
/home/benja/git/libiio/iiod/iiod.c:531: undefined reference to `start_usb_daemon'
[100%] Built target iio_stresstest
collect2: error: ld returned 1 exit status
make[2]: *** [iiod/CMakeFiles/iiod.dir/build.make:227: iiod/iiod] Error 1
make[1]: *** [CMakeFiles/Makefile2:466: iiod/CMakeFiles/iiod.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[100%] Built target iio_attr
make: *** [Makefile:136: all] Error 2

into this:
CM [... cut]
closed 2023-12-07 by pcercuei