analogdevicesinc/libiio #985: libiio 1.0: cannot create new buffer after cancel operation

catkira, 2023-06-14:
When I abort a dequeue operation with `iio_buffer_cancel` and then try to create a new buffer with `iio_device_create_buffer` I get 
```bash
ERROR: Unable to create buffer: Device or resource busy (16)
```
for about 3 seconds. After that `iio_device_create_buffer` works again without doing anything else. Is this a bug? It would be nice if I could create a new buffer immediately after cancel for my application.

catkira, 2023-06-14:
This is the output of iiod when I run it on the device in a terminal
```bash
New client connected from 192.168.137.1
New client connected from 192.168.137.1
New client connected from 192.168.137.1
New client connected from 192.168.137.1
New client connected from 192.168.137.1
New client connected from 192.168.137.1
New client connected from 192.168.137.1
New client connected from 192.168.137.1
New client connected from 192.168.137.1
Client exited
Client exited
Client exited
Client exited
Client exited
Client exited
Client exited
Client exited
Client exited
Client exited
New client connected from 192.168.137.1
Client exited
Client exited
```
for every unsuccessful `iio_device_create_buffer` I get a` New client connected from 192.168.137.1` on the terminal. Then suddenly I get a lot of `Client exited` all at once.

catkira, 2023-06-14:
Ah I figured out that the `Device or resource busy (16)` errors comes after a cancel operation for the duration of the timeout I set with `iio_context_set_timeout`. So I guess that there is probably an active timeout timer running that is not properly reset when 'iio_buffer_cancel' happens.

catkira, 2023-06-14:
Even if I don't do `iio_buffer_cancel` and just let the `iio_block_dequeue` timeout, I have to wait another timeout duration before I can create a new buffer.

catkira, 2023-06-15:
The problem seems to appear only when I run libiio 1.0 on the network client side. local libiio 1.0 and client libiio 0.24 does not show this problem.

catkira, 2023-06-16:
A call to `iio_buffer_disable()` is also blocking for the duration of a timeout. Inside this call is basically only this code
```c
	err = iio_buffer_set_enabled(buffer, false);
	if (err < 0 && err != -ENOSYS)
		return err;

	iio_task_stop(buffer->worker);
```
`iio_task_stop(buffer->worker);` also appears in `iio_buffer_cancel()` so that's probably the problem.

catkira, 2023-06-16:
The wait for the condition in iio_task_stop() is probably blocking
```c
void iio_task_stop(struct iio_task *task)
{
	iio_mutex_lock(task->lock);
	task->running = false;
	iio_cond_signal(task->cond);
	iio_cond_wait(task->cond, task->lock, 0);
	iio_mutex_unlock(task->lock);
}
```

pcercuei, 2023-06-21:
Noted, thanks for testing Libiio v1.0 btw.

pcercuei, 2023-07-11:
@catkira It should be fixed in the PR https://github.com/analogdevicesinc/libiio/pull/1000

pcercuei, 2023-08-24:
Closing as this issue was fixed.

closed 2023-08-24 by pcercuei