analogdevicesinc/libiio #990: OSError: exception: access violation writing 0x0000000000000024

yokihiraHaif98, 2023-06-20:
I'm trying to used Adalm-pluto sdr to capture signals the same script runs correctly from IDLE but when I run it from cmd I get the following error " File "C:\Users\CoSiNC\AppData\Local\Programs\Python\Python310\lib\site-packages\iio.py", line 995, in __del__
OSError: exception: access violation writing 0x0000000000000024"

yokihiraHaif98, 2023-06-21:
The OS: Windows 11, the pylibiio version is 0.23.1. The problem as I mentioned the code works perfectly when I run it from the IDLE and I'm able to capture the signals correctly. The issue happens when I run the file from CMD I get the previous error. I've put all the message error, there is no specific line for that error.

tfcollins, 2023-06-21:
Which line generates the error? Can you also format you code in the post?

tfcollins, 2023-07-05:
Closing due to inactivity. Reopen if still an issue

closed 2023-07-05 by tfcollins