analogdevicesinc/libiio #1025: Pointer NULL checking

oharwot, 2023-08-10:
Hello

there are a lot of pointers in the IIO and almost never they are checked against not NULL.
As an example can be taken function
iio_channel_do_find_attr(const struct iio_channel *chn, const char *name)
in case chn is NULL the function fails.

Would it possible (and does it make sense) to add these NULL checks to all functions?

I am asking as in my setup, due to different bug, the iio crashed at segmentation fault:
1.) the DAC device (/sys/bus/iio/devices/iio:device) was not properly populated
2.) the iii-osciloscope asked for /sys/bus/iio/devices/iio:device3/out_sampling_frequency
3.) this file was not present 
4.) the osc failed at Segmentation fault

I can do the work as well, but I am not the familiar with libiio code so would need to have agreement at format  (e.g. assert or debug printf and return NULL).

Ondrej

pcercuei, 2023-08-10:
Hi Ondrej,

Libiio's API functions will always assume that the pointers passed as argument are valid, so it's your program's responsability to pass valid arguments. In your case, the OSC should be fixed so that it doesn't pass a NULL pointer as argument.

-Paul

oharwot, 2023-08-10:
Hello

OK in this case we can either close or form these assumptions in form of "asserts" :-)

Ondrej

rgetz, 2023-08-11:
I think this has been discussed in the past - and the conclusion has always been segv is OK, and fixing your calling application is the best idea.

oharwot, 2023-08-14:
OK - my opinion is different but lets close.

closed 2023-08-14 by oharwot